trezor/trezor-firmware #1133: wording for "Safety checks" feature

matejcik, 2020-07-24:
in #1126 we introduced "Unsafe prompts" setting. This is confusing because of double-negatives: "enabling unsafe prompts" is the weaker setting and "disabling" is stronger.

We came up with an idea to rename this to "safety checks", with configurable levels, the default being "strict".

For trezorctl, we should change:
* `trezorctl set unsafe-prompts off` to `trezorctl set safety-checks strict`
* `trezorctl set unsafe-prompts on` to `trezorctl set safety-checks prompt`

For protobuf the following field should be changed:
```
optional bool unsafe_prompts = 9;  // allow or disallow unsafe prompts
```
to:
```
enum SafetyCheckLevel {
    STRICT = 0;
    PROMPT = 1;
}

optional SafetyCheckLevel safety_checks = 9;
```
This way binary compatibility holds, because `bool` is basically an enum with `FALSE = 0, TRUE = 1`

Open question: how to reword the on-device screens?

current, for lowering strictness:

> **Unsafe prompts**
> Trezor will allow you to confirm actions which might be dangerous.
> 
> Allow unsafe prompts?

for raising:

> **Unsafe prompts**
> Do you really want to disable unsafe prompts?

andrewkozlik, 2020-08-18:
Another idea for the possible settings. It would be nice to have an "enable temporarily" option. This would enable unsafe prompts until the next reboot, i.e. it wouldn't be written to storage. The typical use-case for this would be when a user has BTC sent to a LTC address and just needs to do a one-off fix. So it decreases the number of steps that the support team needs to instruct the user to do.

tsusanka, 2020-09-02:
Let's agree on this. I like both comments how about we have three stages then: strict, prompt and temporary.

- `strict` is the default setting and does not allow anything
- `prompt` prompts on every such action
- `temp` prompts on every such action but changes to strict on reboot

Wording suggestions:

strict:
```
Safety checks
Trezor will forbid all uncommon actions which might be potentially dangerous.

Are you sure?
```

prompt:
```
Safety checks
Trezor will allow some uncommon actions which might be potentially dangerous.

Are you sure?
```

temp:
```
Safety checks
Trezor will allow some uncommon actions which might be potentially dangerous.
This setting will be revoked after reboot.

Are you sure?
```

The third might be too long and I don't like "uncommon" very much. Ideas?

mmilata, 2020-09-02:
Unusual?

Also not sure about "reboot", might sound like you have to reboot your computer:) Not sure if "unplugging your Trezor" fits on the screen.

andrewkozlik, 2020-09-02:
I would try to be more succinct:
`Trezor will let you approve some actions which might be unsafe.`
or
`Trezor will allow you to approve some actions which might be dangerous.`

andrewkozlik, 2020-09-02:
This fits nicely: `Trezor will allow you to approve some actions which might be unsafe.`
But I am still not satisfied. I would like it to say something like "Do you wish to set safety checks to 'Prompt'?" and then the explanation.
[image]

andrewkozlik, 2020-09-02:
I don't think we need to go into details about unplugging/rebooting, because it makes the explanation too lengthy. Using the word "temporarily" should do:
[image]

tsusanka, 2020-09-02:
I like @andrewkozlik's suggestions, let's go that way! @mmilata feel free to do this one as well, I think @matejcik does not have any work done here yet.

mmilata, 2020-09-02:
:+1:

For future reference there's some discusion regarding the temporary-prompt option in https://github.com/trezor/trezor-firmware/pull/1240#issuecomment-685701950.

tsusanka, 2020-09-25:
I think we can close this one?

matejcik, 2020-09-25:
the wording was changed in https://github.com/trezor/trezor-firmware/pull/1240 it seems

closed 2020-09-25 by tsusanka