trezor/trezor-firmware #1139: Crown Sign Message forbidden key path

ahmedbodi, 2020-07-24:
With an upto date trezor on version 2.3, using the following command: 

trezorctl sign-message -n "m/44'/72'/0'/0/0" "ahoy"

Returns the following error:
Error: DataError: Forbidden key path

Running on OSX using latest python trezorlib

ahmedbodi, 2020-07-24:
Main reason for testing this, I'm working on the crown electrum, and when signing a message it produces a signature, when verifying it, the address that it seems to think the message came from via the public key is different so thats why i decided to test it with trezorctl

ahmedbodi, 2020-07-24:
Figured that part out but it still doesnt work:

trezorctl get-address -c Crown -t address -n "m/44'/72'/0'/0/0" == no address

trezorctl sign-message -n "m/44'/72'/0'/0/0" -c Crown "ahoy"
Please confirm action on your Trezor device.
message: ahoy
address: <null again>
signature: H4Pb55+Uo2Xc2PUGnMksimKMdw1Oo+qsTcrL4UO7sBAbDsuqtzvFZCqfBEmeTrHpmpEsCZXIge30aX+WBaSSukc=

tsusanka, 2020-07-27:
Confirming `trezorctl get-address -c Crown -t address -n "m/44'/72'/0'/0/0"` does not show any address on current master (fdcb64ac2475b2fb738c75e97e6d110673e5b791). No error in emulator log though.

matejcik, 2020-07-27:
this problem must have existed since 6ff644c3045b9d0d9c0327a21f44d208bd92ea21, but nobody noticed for over a year. goes to show how many users use CRW

ahmedbodi, 2020-07-27:
not neccesarily, there wasnt a working electrum wallet that was updated. Its a project ive just taken up. there's lots of users but they all use the core wallet

matejcik, 2020-07-28:
@ahmedbodi @ashotkhachatryan the Crown definition states that `min_address_length` is 36 and `max_address_length` is 40.
Under what circumstances will the address have more than 36 characters?

ashotkhachatryan, 2020-07-28:
@matejcik I just checked. The Crown address will always be 36 characters.

commit 442a7b5f (2020-07-28): core: increase address length limit to accomodate CRW (fixes #1139)
commit 5385617e (2020-08-07): core: increase address length limit to accomodate CRW (fixes #1139)
closed 2020-08-07 by matejcik