trezor/trezor-firmware #1141: FIDO2 site name mismatch

bosomt, 2020-07-27:
**Describe the bug**

I just tried to register new FW version to Github.com.
See screenshots. Device was already registered.
After I removed previous device I saw correct site name.

Even after I tried again to register it and it was was already registered I saw Github.com

**Firmware version and revision**
2.3.2 9bcab40 freeze

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Suite dev master b0d6438272255da85b680ecfa93954f3c10953cf
 - OS: nixos
 - Browser Chromium 
 - Version 84.0.4147.89

**To Reproduce**
Steps to reproduce the behavior:
1. github.com
2. try to add token already registered token
3. check screen

**Expected behavior**
Display will show Github.com

**Screenshots**
[image]
[image]
[image]

bosomt, 2020-07-27:
+ I had issue with Dropbox when registering, new token was registered but device display looked like this till i sent new request to device : 

[image]

andrewkozlik, 2020-07-27:
There are two separate issues:
1. GitHub: I am unable to reproduce this but I have a hypothesis of what happened. GitHub used to have two RPIDs. One for U2F (https://github.com/u2f/trusted_facets) and one for FIDO2 (github.com), now they use github.com for both and it looks like they switched to FIDO2 completely. I think the device was previously registered with GitHub via U2F (using a browser which does not support FIDO2). Now attempt to register as shown above was conducted via FIDO2 and the device showed the old RPID for U2F in the "already registered" dialog. I think the same thing would have happened if you tried to log in using this credential; did you by any chance try that before deleting it @bosomt ? We could improve the user experience by always showing the label of the application "GitHub" instead of the RPID, but if I recall correctly, the spec states that we have to show the RPID in FIDO2, so I don't think we have much choice. A more moderate solution would be to mark the RPIDs which are unique to U2F in knownapps and show the label instead of the RPID in these cases. To determine whether it's worthwhile, it would be good to verify whether the same thing happens during login, i.e. we would need someone with an old U2F credential for GitHub. By old I mean probably a few months or more and by "U2F credential" I mean it was registered using a browser which does not support FIDO2 or using firmware which does not support FIDO2 (any T1 firmware or about 10+ months old TT firmware).
2. Dropbox: I think I have seen this happen occasionally before, I don't think it's specific to Dropbox and I think I only observed it on the device, not the emulator. Looks like the device is forgetting to redraw the homescreen after the workflow is finished. Maybe some race condition. Definitely worth looking into.

bosomt, 2020-07-27:
@andrewkozlik sent that second issue as issue ;) https://github.com/trezor/trezor-firmware/issues/1143

bosomt, 2020-07-27:
@andrewkozlik about that first/master issue , 

i think you are right, I was not able to reproduce it when i removed/added/added second time via Chrome. Maybe I can try to replicate it via Firefox, last time I heard that it doesn't support FIDO2, correct me if I'm wrong. So I can add via Firefox and try add once again via Chromium if needed to confirm this suspicion.

andrewkozlik, 2020-07-27:
> @andrewkozlik sent that second issue as issue ;) #1143

Thanks!

> i think you are right, I was not able to reproduce it when i removed/added/added second time via Chrome. Maybe I can try to replicate it via Firefox, last time I heard that it doesn't support FIDO2, [...]

I tried replicating it via Firefox as you describe, but I wasn't successful, because Github seems to have stopped using "https://github.com/u2f/trusted_facets" for U2F in favor of "github.com".

bosomt, 2020-07-27:
@andrewkozlik I just tried and , yes I was not successful to replicate it via Firefox add and Chromium duplicate add

andrewkozlik, 2020-07-30:
I changed the way we display the application labels so that you should never see something like "https://github.com/u2f/trusted_facets". In U2F we used to translate it to GitHub, but in FIDO2 we didn't translate it. Now I unified the behavior so that it will always show up as "github.com" everywhere. The same applies to all the other U2F applications, e.g. instead of "Slush Pool" users will always be seeing "slushpool.com", so that we are consistent in always showing the URL origin.

@qa Test this with several sites like GitHub, Binance, DropBox or anything else you like. In each case:
1. Register using U2F (using some version of Firefox that doesn't support FIDO2 yet). Try logging in.
2. Register using FIDO2 (using Chrome/Chromium). Try logging in.

Ensure that in both cases you see the same label.

closed 2020-07-30 by andrewkozlik

bosomt, 2020-08-12:
2.3.2 63ebb8ccb

Dropbox - Register U2F, Login FIDO2 - Logo + Dropbox 
GitHub   - in progress 
Binance - in progress