trezor/trezor-firmware #1142: Unable to run device tests on TT

tsusanka, 2020-07-27:
**Describe the bug**
	Unable to run device tests on TT. Trezor does not answer DebugLinkGetState message.

	**Firmware version and revision**
	Current master (8dacad0).

	**To Reproduce**
	Steps to reproduce the behavior:
	1. Connect Trezor T with debug mode enabled.
	2. `pytest -vs tests/device_tests/ -k test_reset_device_slip39_advanced`

	**Additional Context**
	Full log: https://gist.github.com/tsusanka/3b237603ea2c2ab9fd65e2e4b48e75b4

	The most interesting part is the end obviously. The `DebugLinkGetState` is not answered.

	```
	[2020-07-27 10:02:31,040] trezorlib.debuglink DEBUG: sending message: DebugLinkDecision
	DebugLinkDecision (11 bytes) {
	    input: 'ancient',
	    wait: False,
	}
	[2020-07-27 10:02:31,042] trezorlib.debuglink DEBUG: sending message: DebugLinkGetState
	DebugLinkGetState (2 bytes) {
	    wait_word_pos: True,
	}
	```

	I guess this is some kind of race-condition error. @matejcik could you have a look?

tsusanka, 2020-07-27:
@mmilata wrote:

```
Parkrat jsem je spustil se stejnym vysledkem. Konkretne test_msg_resetdevice_slip39_advanced.py::TestMsgResetDeviceT2::test_reset_device_slip39_advanced se na necem sekne i kdyz ho pustim samotny.

Podle git bisect je asi problem v  85d74ece (edited) 

Hm tak ne, s predchozim commitem mi samotny test prochazi ale kdyz ho pustim se vsema ostatnima zustane viset ... (edited) 
```

mmilata, 2020-07-27:
When running all device tests in random order, I've seen getting stuck on:

```
test_reset_recovery_slip39_advanced.py::test_reset_recovery
test_msg_backup_device.py::test_backup_slip39_advanced
test_msg_resetdevice_slip39_advanced.py::TestMsgResetDeviceT2::test_reset_device_slip39_advanced
test_reset_backup.py::test_skip_backup_msg[2-backup_flow_slip39_advanced]
test_reset_backup.py::test_skip_backup_manual[2-backup_flow_slip39_advanced]
```

matejcik, 2020-07-28:
got a stack trace:
```
Traceback (most recent call last):
  File "trezor/wire/__init__.py", line 369, in handle_session
  File "trezor/loop.py", line 526, in __iter__
  File "trezor/loop.py", line 187, in _step
  File "apps/management/backup_device.py", line 21, in backup_device
  File "apps/management/reset_device/__init__.py", line 194, in backup_seed
  File "apps/management/reset_device/__init__.py", line 151, in backup_slip39_advanced
  File "apps/management/reset_device/layout.py", line 503, in slip39_advanced_show_and_confirm_shares
  File "apps/management/reset_device/layout.py", line 135, in _show_share_words
  File "apps/common/confirm.py", line 114, in require_hold_to_confirm
  File "apps/common/confirm.py", line 104, in hold_to_confirm
  File "trezor/loop.py", line 347, in __iter__
  File "trezor/loop.py", line 187, in _step
  File "trezor/ui/__init__.py", line 334, in __iter__
  File "trezor/loop.py", line 347, in __iter__
  File "trezor/loop.py", line 191, in _step
  File "trezor/ui/__init__.py", line 380, in handle_rendering
  File "apps/debug/__init__.py", line 53, in notify_layout_change
  File "trezor/loop.py", line 433, in publish
MemoryError: memory allocation failed, allocating 2048 bytes
```

matejcik, 2020-07-28:
The problem is that by default the test suite enables layout recording, and most test cases don't actually read the layouts.

So in testcases with many screens, as listed above, the list might grow too big and the RAM too fragmented to handle the next step increase.
(2048 bytes matches a list that has grown beyond the allocated size of 1024 bytes = 256 items)

i'm not entirely convinced that this is all there is to it, but it's something that needs fixing in any case

tsusanka, 2020-07-29:
@matejcik I am still unable to run the whole test suite on device. Recovery is just fine but `test_msg_webauthn.py` is failing for me even when run on its own using `pytest -s tests/device_tests/test_msg_webauthn.py`. Haven't we discussed we will lower the number of allowed residential credentials?

Tested on b6ec3de72a81615ef2bbae1519897f335f5106b5

tsusanka, 2020-07-29:
Works fine with 75759cba943b57107ed4a382bd3bb5a20bba3ff1 as suspected but we need to discuss it obviously.

matejcik, 2020-08-05:
this should be fixed in #1168

tsusanka, 2020-08-05:
I was able to run regular device tests on an actual device on release/2020-08 branch (cde8f31ec2ddcb7d35e36edbcf8a71dda983a9ea) with d776cc931bf8b412ac8af7eae709efc584f7b0cd and 741fca01567151fa9dad2e9d25db7236169fb697 cherry-picked. 

The test suite was halted twice on some screens (test_msg_eos_signtx.py and bcash signtx). After I have confirmed the dialogs manually the test suite has continued. 

Four tests have failed:

- test_softlock_instability
- TestMultisig.test_2_of_3
- TestMultisig.test_15_of_15
- TestOpReturn.test_opreturn 

`test_softlock_instability` and `test_2_of_3` fail also when run on its own.
`test_15_of_15` and `test_opreturn` succeed every time when run on its own and might have been caused by the "manual" click.

---

In case these issues are due to some weird race conditions we can fix it after #946 is done to make it easier. However, the first two tests are making me a bit nervous since they fail deterministically (on a first look).

matejcik, 2020-08-06:
`test_softlock_instability` is expected to fail on device because `reseed()` does not exist in hardware builds. Let's not change that, but if we had a way to determine that the test is running on a physical hardware, we can xfail it.

matejcik, 2020-08-06:
I was unable to reproduce the `test_2_of_3` failure so far. Neither the others, but I have not finished the full suite yet.

tsusanka, 2020-08-06:
Hm I am unable to reproduce `test_2_of_3` again neither on master nor on the release branch (plus fixes) :/.

tsusanka, 2020-08-17:
I was unable to reproduce this anymore. Let's finish #946 first and see how that goes in CI.

commit b6ec3de7 (2020-07-28): tests: watch_layout must be explicit (fixes #1142)
PR #1147 (2020-07-29): use watch_layout only explicitly
fixes #1142 (hopefully), by not saving all layouts in memory when nothing is reading them
commit 741fca01 (2020-07-29): tests: watch_layout must be explicit (fixes #1142)
closed 2020-08-17 by tsusanka