trezor/trezor-firmware #1143: Occasional redraw the homescreen forgotten in latest master

bosomt, 2020-07-27:
**Describe the bug**
Sometimes it happens that device display or suite or some U2F/FIDO2 login dialgoue
will stay opened after I confirm unlock PIN dialogue.
I will add more information during test of new 4eb5b927 revision

**Firmware version and revision**
2.3.2 9bcab40 freeze
and white noise background

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Suite dev master b0d6438272255da85b680ecfa93954f3c10953cf
 - OS: nixos
 - Browser Chromium 
 - Version 84.0.4147.89

**To Reproduce**
Steps to reproduce the behavior:
1. Suite
2. wait for autolock
3. go to settings and add other coins
4. unlock device
5. see that accounts are loaded but device PIN dialogue is still overlapping Suite accounts 

**Expected behavior**
Dialogue should disappear from Suite and device
**Screenshots**

**Add accounts in suite**
[image]
**Add security token in Dropbox security**
[image]

tsusanka, 2020-08-06:
@bosomt can you still reproduce this on the released 2.3.2? And then on current master?

bosomt, 2020-08-06:
@tsusanka i was not able to reproduce it with 2.3.2 release.
 i'm still testing new FW with new suite, 
I will close it in few hours if i won't be able to reproduce it

bosomt, 2020-08-07:
i was not able to reproduce it with 2.3.2 63ebb8ccb

closed 2020-08-07 by bosomt