trezor/trezor-firmware #1148: Double PIN entry when disabling PIN after autolock timeout

sorooris, 2020-07-28:
**Describe the bug**
PIN entry enforced twice before disabling it in Wallet; to unlock the device after autolock timeout and then again after PIN disabling confirmation dialog on device.

**Firmware version and revision**
tested 2.3.2 regular 4eb5b92

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Mytrezor 0c93d2e64b0889b713d875a9483955fd2a71d12a
 - OS: any (tested NixOS)
 - Browser: any (tested Chromium 83.0.4103.116)

**To Reproduce**
Steps to reproduce the behavior:
1) session timeout - device locked
2) enter PIN to unlock device 
3) confirm PIN disabling dialog
4) enter PIN again

**Expected behavior**
After unlocking the device for session the PIN is not asked again after disabling.

**Additional context**
I am very unsure of the expected behavior beause it could open a potential vulnerability so I'd like to confirm that this is what we expect.

tsusanka, 2020-07-28:
While you are properly right this may seem redundant I believe asking for PIN again makes sense becuase turning off PIN is a big thing. I don't think it's worth the hassle to worry about this.

sorooris, 2020-07-28:
okay, thanks for confirming - closing.

tsusanka, 2020-07-28:
But let's also see what other think - @prusnak @matejcik @andrewkozlik.

andrewkozlik, 2020-07-28:
1. From a UX perspective I am not happy about this.
2. From a programmer perspective I see two cases:
    (a) Changing or disabling PIN or the wipe code: This doesn't seem worth the hassle to improve the UX, because we need to provide the correct PIN to the `change_pin()` C-function. That means we would need to carry the PIN entered in the lockscreen over to the change PIN workflow. Probably not difficult but seems a bit messy. Maybe @matejcik can see an elegant way to do this.
    (b) Other PIN entry requests which are intended for user verification only. Here I think it is worth the hassle. A typical example is FIDO2, where I already solved this. Another example would be dry run recovery, where this probably isn't solved and I think it should be. Can anybody recall any other places where we check the PIN?

@sorooris just check that for FIDO2 you don't see this happening and try dry run recovery to see what happens.

matejcik, 2020-07-29:
re (a) we could use something akin to passing the preauth object -- either for a fixed set of messages, or make all handlers accept a "preauthorization" argument.

In fact this might be something to add to the `Context` object? Whose role is rather unclear at this point. I'd expect the `Context` to be bound to a session in the future, maybe we could start working on it now? It would require some thought though.

re (b) we could reuse the "timeout" mechanism from FIDO

andrewkozlik, 2020-07-29:
> In fact this might be something to add to the `Context` object? Whose role is rather unclear at this point. I'd expect the `Context` to be bound to a session in the future, maybe we could start working on it now? It would require some thought though.

Sounds sensible. However, I wouldn't want the PIN lying around there longer than it has to and storing the PIN in the `Context` would have to be restricted to a subset of messages (`ChangePin`, `ChangeWipeCode`, `SdProtect`).

sorooris, 2020-07-29:
@andrewkozlik 
Device: 2.3.2 regular 4eb5b92
Mytrezor wallet: develop branch, commit 0c93d2e64b0889b713d875a9483955fd2a71d12a
- FIDO2 - tested on [webauthn](https://webauthn.io/) and PIN is asked only once for unlocking the device for both registration and login.
- Dry-run:

1. PIN dialog is displayed (to unlock device)
2. Dry-run confirmation screen is displayed
3. **PIN dialog is displayed again**

Then the Seed check - number of words selection screen is displayed.

tsusanka, 2020-09-17:
We have discussed this in Product and we do not think this is problematic but rather the other way around. Requesting PIN twice for such a sensitive operations (such as PIN disable or dry run) seems just fine.

closed 2020-09-17 by tsusanka