trezor/trezor-firmware #1154: Implement autolock as a "screensaver"

tsusanka, 2020-07-31:
As a followup of #1099 we have discussed the autolock acts problematically when the user is doing a backup/recovery. For example in case the user has a very low autolock delay (10s) and they read the mnemonic words longer than the 10s (which sounds reasonable) the device autolocks itself and completely aborts the backup process. What's even worse is that since the backup is aborted it can not be done again.

matejcik, 2020-08-24:
We currently don't know how this would work internally. Updated issue name to reflect that the current step is research.

tsusanka, 2020-11-18:
Upon discussion with @matejzak we have agreed this is not worth it. Thanks @matejcik for bringing this up. We will solve the edge case of interrupting backup/whatever by raising the minimum autolock value from 10 sec to 1 minute. We believe that's the most straightforward solution.

If we gather some user feedback that the value is too low / too high we will reevaluate.

closed 2020-11-18 by tsusanka

matejcik, 2023-04-12:
> as a tangent, would it be an option to pause the auto-lock timer for the duration of essential tasks like passphrase entry and transaction approval?

This is precisely the thing that we _don't_ want to do: waiting for user input indefinitely means that if, i dunno, you notice your dog running into traffic while you're signing, anyone can come back to confirm the thing you possibly didn't want.

In #2651 we made sure that softlock will not interrupt Bitcoin signing _after_ you confirmed it, but it _will_ interrupt while waiting for confirmation. This does not seem at all appropriate here; besides the "screensaver-like" behavior that lets you go back to where you left off, the only viable option is to artificially bump the autolock timer by a minute or two when you're running an essential task.