trezor/trezor-firmware #1156: Connectivity error in Monero wallet GUI when TT is locked

wendys-cats, 2020-08-02:
**Describe the bug**
When creating OR entering wallet, Monero-wallet-GUI responds with errors (Device connect failed). This occurs exclusively when auto-lock-delay feature is set, and the device is locked at the time of request. Seems that Monero-wallet-GUI either checks for device in a similar fashion to trezorctl (i.e. will respond with "Failed to find a trezor device" if it is locked), or firmware failed to wake up the device. 

**Firmware version and revision**
Applies for TT device, tested on 2.3.2, revisions 731e9b7c and 4eb5b92

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Monero wallet GUI (tried Nixos 0.15.4. version, and Windows 0.16.0.2 version
 - OS: nixos and windows 10

**Steps to reproduce the behavior:**
1. Set up Monero wallet GUI
2. Set up TT device with PIN and auto-lock-delay 
3. Create OR enter wallet in Monero wallet GUI when device is locked

**Expected behavior**
Expected for device to wake up and see "Enter PIN" screen.

tsusanka, 2020-08-03:
@ph4r05 is this something you might have a look at when you have a bit of time? We have added an auto-lock feature for TT and during that we have added a new Button Request "PinEntry" which denotes that PIN is being requested.

bosomt, 2020-08-04:
I was not able to reproduce it with same Win10 0.16.0.2 version
FW 2.3.2 revision 731e9b7
/not like Electrum where it happens everytime/

tsusanka, 2020-08-05:
Could @bosomt discuss with @wendys-cats whether this is worth investigating or not? Unless there is a clear path to reproduce the error I am not sure it makes sense.

tsusanka, 2020-08-18:
@bosomt @wendys-cats ping. Were you able to reproduce this?

bosomt, 2020-08-18:
i'm sorry but my monero 0.16.0.2 doesn't work even when device is unlocked
so please wait till it try to reproduce it in Linux or Windows and not in nixos

2020-08-18 18:58:25.708	I Failed to invoke http request to  /json_rpc
2020-08-18 18:58:25.708	E Failed to connect to daemon
2020-08-18 18:58:25.710	I Failed to invoke http request to  /json_rpc
2020-08-18 18:58:25.710	E Failed to connect to daemon
2020-08-18 18:58:25.710	E Device not found in registry: 'Trezor'. Known devices: 
2020-08-18 18:58:25.710	E  - Ledger
2020-08-18 18:58:25.710	E  - default
2020-08-18 18:58:25.721	I ~WalletImpl
2020-08-18 18:58:25.721	I closing wallet...
2020-08-18 18:58:25.721	I Calling wallet::stop...
2020-08-18 18:58:25.721	I wallet::stop done
2020-08-18 18:58:25.722	I ~WalletImpl finished

[image]

wendys-cats, 2020-08-20:
@tsusanka on Monero GUI 0.16.0.2 on Windows 10 wallet creation consistently results in "Device conncet failed" for me. Will also try newer version (0.16.0.3) and let you know.

ph4r05, 2020-08-20:
Interesting, according to the logs from @bosomt it looks that Trezor support is not compiled in.

bosomt, 2020-08-20:
after uninstall of 0.16.0.2 install of 0.16.0.0 version  it's working again.

probably id depends if it's in LOCKED or NOT CONNECTED lock state.

when i try to open wallet and device is only autlocked i will get correct PIn unlock dialogue 

[image]

same result when trying to create wallet

[image]

of course I will get error when device is not connected state

ph4r05, 2020-09-04:
@tsusanka regarding the auto-lock and the `PinEntry`, do you think the Trezor client in Monero needs some change? I would have a few minutes for this (as I recently did the CLSAG PRs).

tsusanka, 2020-09-04:
@ph4r05 it depends on how Monero behaves to unknown ButtonRequestTypes. If it simply does not care and sends ButtonAck then it should not require anything. If it dies on "unkown protobuf field" or something like that then we need some modifications. cc @matejcik

ph4r05, 2020-09-04:
@tsusanka I've just checked, it seems that message type for `PinEntry` is `MessageType_ButtonRequest`. Client does not care about unknown ButtonRequest codes, so if no reaction from our side is needed (besides sending the `ButtonAck`) then no modification is needed.

tsusanka, 2020-09-04:
@bosomt could you do a final test of this? And if you find anything reproduce it into something sensible. Otherwise it is really hard to grasp what the problem could be

tsusanka, 2021-03-18:
We have not heard about this since so we are closing.

closed 2021-03-18 by tsusanka