trezor/trezor-firmware #1159: Do we print coin name during message signing?

tsusanka, 2020-08-03:
We should print the Coin name during message signing so the attacker can not trick the user into signing a message with a different coin than intended.

tsusanka, 2020-08-03:
@mmilata do you think you could have a look please?

closed 2020-08-18 by tsusanka

sorooris, 2020-08-24:
**QA OK**
Device: model T 2.3.3 e137f196, model 1 1.9.3 e137f196
Wallet: public [ec6b4d](https://github.com/satoshilabs/mytrezor/commits/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0)
Tested with BTC, LTC and ZEC; device correctly displays the coin's name in the screen header when signing (the single screen) and verifying (both screens).

bosomt, 2020-08-26:
**QA OK**

model T revision: df5421e
wallet ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0
tested with BTC,DOGE,ETH,VTC

sorooris, 2020-08-26:
**QA NOK**
Device: model 1 1.9.3 e137f196, df5421e7
Wallet: public [ec6b4d](https://github.com/satoshilabs/mytrezor/commits/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0)
Missing coin name in model 1.

[image]

tsusanka, 2020-08-26:
I am suggesting not to do this and solve by _Port_.

bosomt, 2020-12-21:
**QA OK**

model One is missing Coin information as expected https://github.com/trezor/trezor-firmware/issues/1159#issuecomment-680845515

Nixos
wallet.trezor.io be4e63a6db60aaa07ae982fd77262c59a686ad88
Device: fw-regular-2.3.5-889a3699 + trezor-fw-regular-1.9.4-889a3699

PR #1181 (2020-08-18): core: display coin name when signing message
Fixes #1159. Also displays the coin shortname when showing address during the `verify_message` flow.

[UI diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/681119460/artifacts/master_diff/index.html)