trezor/trezor-firmware #1167: Consider 18-digit PIN

prusnak, 2020-08-05:
This was implemented by exeabyss in https://github.com/trezor/trezor-firmware/pull/721 - more particularly in commit  https://github.com/trezor/trezor-firmware/commit/45534168a75f3b59d72e28a0b146b583019a0a34

The commit needs some love and implementation for Model T is missing, but we need to address the master question first:

Do we want to increase the maximum length of the PIN from 9-digits to 18-digits?

HeroHann, 2020-08-06:
I don't know if I am even allowed to comment on this...anyway:

IF there is a security advantage with more digits AND the user can still choose to only use 6 or 9 or 12 digits. Why not?

andrewkozlik, 2020-08-12:
Correct me if I am wrong, but as I recall @tsusanka told me several months ago that he discussed this at one of the product meetings and the decision was to go ahead and implement unlimited PIN length. So I think we can remove the "Product Owner" label. BTW, by "unlimited" I mean that the solution should not introduce any hard limitations on the length, like we have now. Of course we will set some limit, but it will be easy to change in the future.

The biggest problems here are that it will require:
- An update to the way we compute the KEK and KEIV in the storage, i.e. most likely a storage format upgrade, which is what stopped the original PR in its tracks.
- An update to the way we check the wipe code.

For the KEK computation there are these options:
1. When checking or setting the PIN support both the old and the new computation method and decide between the two based on the length of the entered PIN. No storage upgrade is required, but it's messy forcing us to keep legacy code at the heart of our codebase.
2. When the user unlocks their Trezor for the first time after the firmware update, upgrade the storage format from the old computation method to the new one. When checking the PIN, we still have to support both methods, but for setting the PIN, only the new method. After a few years we can remove the old method completely and require any dormant users to do a two-step firmware upgrade.

The wipe code cannot be upgraded. There are these options:
1. Same as Option 1 for the KEK.
2. When setting the wipe code support only the new computation method. When checking the wipe code support both the old and the new method and try both every time. If one of them results in a match then wipe the device.
3. When the user unlocks their Trezor for the first time after the firmware update, check whether the user configured a wipe code and if they did, then clear the wipe code and show a message telling the user to configure it again.
4. Keep the wipe code restricted to 9 digits.

We discussed the options during the standup and in case of the KEK we are in favor of Option 2, in case of the wipe code we are also in favor of Option 2, but Option 3 is also worth consideration.

matejcik, 2020-08-12:
> 2\. When the user unlocks their Trezor for the first time after the firmware update, upgrade the storage format from the old computation method to the new one. When checking the PIN, we still have [...]

I assumed we could see storage version before unlock, which tells us which method to use for decoding.

There are two "versions", right? The "field format version" and the "norcow version". I assumed we would have to bump the norcow version, is that not the case? Can the low-level format stay the same?

matejcik, 2020-08-12:
btw perhaps this is another reason to `public`ize the version field?

andrewkozlik, 2020-08-12:
> There are two "versions", right? The "field format version" and the "norcow version". I assumed we would have to bump the norcow version, is that not the case? Can the low-level format stay the same?

I am not sure what you mean by the "field format version". There is the norcow version, which is used by norcow.c and storage.c. It's written in the storage twice, once in plaintext and once encrypted+authenticated to protect against downgrade attacks. This is the version that will get bumped. The format will appear the same but some values will be computed differently. No other module should be interested in knowing what this version is. The modules which build on top of trezor-storage, like trezor-core have their own version number which is independent of the norcow version.

tsusanka, 2020-08-18:
**Product stance:** yes, product agrees with this feature as low priority. Since this requires storage upgrade though and therefore disables downgrading we suggest not to include this in the two upcoming releases (neither 2020-09 nor 2020-10). There were number of bugs/features in the last FWs where the users had to downgrade for some reasons (derivation paths, Zcash). So we would like to merge this after the dust settles a bit. The ideal case would be if there is some other change that requires such measure (disabling downgrade).

andrewkozlik, 2020-08-18:
> There were number of bugs/features in the last FWs where the users had to downgrade for some reasons (derivation paths, Zcash). So we would like to merge this after the dust settles a bit.

Presumably the next firmware release will have those bugs fixed, so I don't see that as a reason to wait. As for people sending their BTC to LTC addresses, that won't be fixed by waiting either. I think what needs to be done as a priority is to add the unsafe-prompts setting to the Wallet + Suite, so that it can be used even by less technical users, or to enable unsafe prompts (perhaps with one more warning screen) until the setting is added.

On a separate note, this feature can go into a beta release.

andrewkozlik, 2020-08-18:
> I don't know if I am even allowed to comment on this...anyway:
> 
> IF there is a security advantage with more digits AND the user can still choose to only use 6 or 9 or 12 digits. Why not?

@HeroHann of course you are allowed to comment. For the record, the security advantage here is rather small. We maintain that the right solution is to use a strong passphrase, because that protects both your Trezor **and** your paper backup, whereas the PIN only protects your Trezor. That being said, a longer PIN has the advantage of providing stronger protection of FIDO2/U2F credentials, which can't be achieved through the passphrase.

tsusanka, 2021-04-16:
@andrewkozlik should we close this one? Since #1478 is merged

closed 2021-04-16 by andrewkozlik