trezor/trezor-firmware #1177: Confusing Version Group ID error message

prusnak, 2020-08-10:
Log from the wallet:

```
...Sending","TxAck",{"tx":{"version":1,"lock_time":0,"inputs_cnt":1,"outputs_cnt":2,"version_group_id":0}}]
...Received","Failure",{"code":"Failure_DataError","message":"Version group ID must be set when version >= 3."}]
```

The following check, while technically correct, spits confusing error message when `version<3` and `version_group_id==0` is provided:

https://github.com/trezor/trezor-firmware/blob/1e58c86a9af9b891abe13fad7866f5852786941f/legacy/firmware/signing.c#L1336

prusnak, 2020-08-10:
There is a pending fix for trezor.js here: https://github.com/trezor/trezor.js/pull/113

closed 2020-09-04 by tsusanka

tsusanka, 2020-09-25:
## QA

Try ZCASH signing please on Wallet and Suite. Notice you exceptionally need to use master (59b8a5a) not what we have frozen.

tsusanka, 2020-12-28:
Zcash tested as part of _Freeze_. See [testpad](https://satoshilabs.ontestpad.com/script/266#6/3/).

PR #1250 (2020-09-04): legacy: replace confusing Version group ID error message
Fixes #1177.