trezor/trezor-firmware #1183: CI test dependencies

andrewkozlik, 2020-08-14:
It might be useful to configure the CI dependencies so that the core tests won't wait for the legacy build to succeed and vice versa. It's not a big deal, but it would be convenient when you make changes in something like storage or crypto, which break the legacy and core builds. After fixing one of the builds you might want to run the test suite, before starting work on fixing the other build. Let's discuss the available options and the rationale behind the current setup.

matejcik, 2020-08-14:
for reference: https://docs.gitlab.com/ee/ci/yaml/#needs should be available to us

prusnak, 2020-08-14:
Yes, I agree that using a DAG using `needs` much better suits our needs.

tsusanka, 2020-08-17:
Great! Replacing `dependencies` with `needs` seem to work just like that :scream:. See this [job](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/178865122/dag). I will check tomorrow if this does not break anything else and will merge then.

tsusanka, 2020-08-17:
Btw if we decide to use this it means that if one job fails the other will continue in case they need not depend on it using `need`. So for example if `style` fails the rest of it still [runs](https://gitlab.com/satoshilabs/trezor/trezor-firmware/-/pipelines/178869273). But I think that is a feature not a bug?

PR #1195 (2020-08-18): Ci: replace 'dependends' with 'needs'
AFAICT this does not break anything.

Closes #1183.
closed 2020-08-18 by tsusanka