trezor/trezor-firmware #1186: Implement SD card protection analugue for T1

yura-pakhuchiy, 2020-08-15:
SD card protection is a great measure against physical seed extraction on TT. Unfortunately there is no SD card on T1. I propose to implement similar feature on T1, but user have to submit secret via trezorctl or copy-n-paste it to a web interface. The secret can be stored in a standard location and python-trezor and/or trezor-bridge will automatically submit it to the device. Or user can move it to a different location and then manually copy-n-paste the secret. This will provide protection comparable to SD card protection if user encrypts his HDD. For TT it would be great to allow both protections enabled at the same time (this and sd-card protection).

andrewkozlik, 2020-08-18:
That's looks like a very neat idea, however here are some of my thoughts:
1. From a usability perspective makes sense on T1, because T1 already needs to be connected Wallet/trezorctl to enter the PIN.
2. Seems relatively easy to implement in T1 firmware, just reuse the salt parameter in storage.
3. Makes less sense on TT:
    - There is already the SD card.
    - We are aiming to make the device less dependent on the computer, e.g. to work in airgapped mode, also we want features like FIDO2/U2F to work on computers without Wallet/trezorctl/bridge. This would be going in the opposite direction.
    - In the future we might be able to integrate Trezor with OSs to login and decrypt the HDD. In that case it becomes a chicken and the egg problem.
    - Trezor T connects to the computer only after unlocking, we would need to connect before unlocking to get the secret. Not sure how insistent we are on this behavior.
4. Most importantly, the passphrase feature can be used to achieve the same result. Let *S* be a high-entropy secret string stored in a text file on your desktop a let *P* be a secret phrase you remember. Configure your passphrase to be *S* followed by *P*. Usability is not as great as it would be if bridge did this in the background, because you have to copy-and-paste *S* manually, but still pretty easy. So I am not sure whether it's worth investing design, development and maintenance into a new niche feature for a minor improvement in usability.
5. Another way to achieve similar security is to have a 2-of-2 multisig account, where one signature is controlled by Trezor and the other by a desktop wallet. It doesn't even have to be classic multisig, it could be 2P-ECDSA, which makes it look like singlesig. It might be a better time investment to integrate a feature like this seamlessly into our desktop wallet software.
6. A major advantage I see in the passphrase solution or the multisig solution is that both of these solutions allow the user to have:
    - A high-value savings account, which would use the superior security and would work only on a computer where the user has the secret *S* stored or where they have the desktop wallet installed.
    - A low-value day-to-day account which wouldn't use the feature and would work on any computer.

    This is not possible with the proposed idea.

yura-pakhuchiy, 2020-08-19:
@andrewkozlik thank you for the detailed answer! Regarding passphrase and multisig, I agree that they provide similar security, but they have following drawbacks:
1. User have to backup passphrase and/or additional seed for multisig. The funds will be lost if they fail to do it. But for sd-card like protection backup is not required and funds can be restored with original seed only.
2. User have to transfer funds to the new seed. Might be not trivial task if user has many UTXO which they do not want to mix. Or if they store many different altcoins.
So sd-card like protection have advantage that user can quickly apply it to existing seed. I think all 3 protections (multisig, passphrase, sd-card like) have their use cases.

BTW, would love to see multisig support (especially 2P-ECDSA) in the official wallet.

matejcik, 2020-08-24:
related (solutions would be similar): #32

tsusanka, 2020-09-17:
Thank you for your valuable suggestion @yura-pakhuchiy! We have discussed this internally and we would like to achieve this via #32 and #79 preferably. Thanks for your feedback!

closed 2020-09-17 by tsusanka