trezor/trezor-firmware #1187: Trezor now rejects "invalid" derivation path for BCH wallets

deadalnix, 2020-08-15:
**Describe the bug**
BCH wallet created with nonstandard - but nevertheless common in practice - derivation path do not function properly on trezor anymore. For instance, electron cash uses 0/45'/0 as a default derivation path for multisig.

**Firmware version and revision**
Lastest.

**To Reproduce**
Create a multisig wallet on electron cash, or use an existing one created with efault parameters. Trezor now refuses to sign transactions.

**Expected behavior**
Transaction are signed after validation on the HW device.

**Additional context**
While I understand that these path are non standard, and that is a pain in the ass to support, they are nevertheless common in practice due to being the default on popular wallets such as electron cash. I personally have several wallets which are affected, and can point to several other Trezor users who are affected.

Maybe an alternative could be to prompt a warning on the trezor wallet itself about the wallet not using a standard derivation path.

prusnak, 2020-08-15:
This is an Electron-Cash bug which caused people to store their coins on invalid paths - see https://github.com/Electron-Cash/Electron-Cash/issues/1963

If you need to transact before the issue is fixed in Electron-Cash you can downgrade the firmware.

closed 2020-08-15 by prusnak

deadalnix, 2020-08-15:
Hi @prusnak ,

With all due respect, people have real money in these wallets. Downgrading the firmware is a nice temporary solution, but not a sustainable one.

prusnak, 2020-08-15:
We'll address this in https://github.com/trezor/trezor-firmware/issues/1188

deadalnix, 2020-08-15:
Fantastic. Thanks.