trezor/trezor-firmware #1192: Increase coins fee_thresholds

tsusanka, 2020-08-17:
As a follow-up of #1087 @mmilata made a great research with current fee threshold values of Bitcoin-like coins.

For some of the bitcoin-based coins, the threshold for 250B transaction is following:

| Coin | `maxfee_kb` | `fee_threshold`        | USD rate | `fee_threshold` (USD) | hard limit (USD)         |
| :--- | ----------: | ---------------------: | -------: | --------------------: | -----------------------: |
| BTC  | 2000000     | 500000 sats = 0.005    | 11500.0    | 57.5                  | 575                      |
| BCH  | 500000      | 125000 sats = 0.00125  | 280.0      | 0.4                  | 4.0                      |
| BTG  | 500000      | 125000 sats = 0.00125  | 10.0       | 0.01                | 0.1                    |
| DASH | 100000      | 25000 sats = 0.00025   | 90.0       | 0.02                | 0.2                    |
| DGB  | 500000      | 125000 sats = 0.00125  | 0.03     | 0.00004             | 0.0004                 |
| DOGE | 1000000000  | 250000000 sats = 2.5   | 0.003    | 0.008                | 0.08                    |
| LTC  | 40000000    | 10000000 sats = 0.1    | 55.0       | 5.5                   | 55.0                       |
| NMC  | 10000000    | 2500000 sats = 0.025   | 0.4      | 0.01                  | 0.1                      |
| VTC  | 40000000    | 10000000 sats = 0.1    | 0.3      | 0.03                  | 0.3                      |
| ZEC  | 1000000     | 250000 sats = 0.0025   | 85.0       | 0.2                | 2.0                    |

----

Product suggests to set `fee_threshold` to $10 (and hard limit to $100 then) for all non-Bitcoin coins. For Bitcoin I would like to see some research and maybe we could set it to fivefold the average fee in the last year?

We brought up those numbers without any proper research whatsoever so any feedback is welcomed though. Let's discuss this in Triage.

mmilata, 2020-08-23:
@tsusanka sure

prusnak, 2020-08-24:
> Is there any hard opposition to this?

I am OK with changing the fees for altcoins. Don't change the fee values for Bitcoin.

tsusanka, 2020-08-24:
@mmilata could you please add Testnet into the table? Obviously without USD columns but it would be helpful to know the threshold also for Testnet.

mmilata, 2020-08-24:
@tsusanka added

tsusanka, 2020-08-24:
So @sorooris if you send between 0.025 and 0.25 TESTNET fee you should get a warning, if you send over 0.25 you should get an error.

closed 2020-08-25 by tsusanka

sorooris, 2020-09-01:
**QA OK**
2.3.3 df5421e7
Suite [3095a174f258c0226faf2ebcbdda3d8b7f04686f](https://github.com/trezor/trezor-suite/commits/3095a174f258c0226faf2ebcbdda3d8b7f04686f)
[image]
[image]
0.18400039 TEST | 0.25551594 TEST
------------ | -------------
[image] | [image]

Wallet [ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0](https://github.com/satoshilabs/mytrezor/commits/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0)
Final fee: 100139.73493173493 sat/B 
Size: 3.003 kB 
Fee: 3.00719624 TEST
[image]