trezor/trezor-firmware #1196: Setting homescreen using trezorctl fails

tsusanka, 2020-08-18:
**Describe the bug**
Setting homescreen using trezorctl fails.

**Firmware version and revision**
FW: 1.9.2
Trezorctl: master (927ee08)

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezorctl
 - OS: NixOS

**To Reproduce**
Steps to reproduce the behavior:
1. `trezorctl set homescreen -f [PATH]`
2. See error: `Error: Invalid value for '-f' / '--filename': /home/tsusanka/documents/redesign/t1/new/12.png is not a valid boolean`

Omitting `-f` does not help. Am I using it right?

Fails on master (927ee08) and `python/v0.12.1`. Works on `python/v0.12.0`.

matejcik, 2020-08-18:
looks like a Click bug
as a quick fix reverting https://github.com/trezor/trezor-firmware/commit/2678e64a9923e56345054daf6eb87ba17b9d3169 should help, i'll look into it in more detail when i get back

mmilata, 2020-08-18:
Ran into this today as well, here's possible fix: #1199.