trezor/trezor-firmware #1203: Enabling extra_data for Stakenet

AlexITC, 2020-08-19:
At Stakenet, we are doing a hardfork to get the latest Dash features for handling masternodes, which ends up supporting [DIP-0002](https://github.com/dashpay/dips/blob/master/dip-0002.md).

Before doing that, we are interested to know if it's possible to get the `extra_data` config enabled for Stakenet, from what I see there are two places to enabled this ([first](https://github.com/trezor/trezor-firmware/blob/master/common/defs/bitcoin/stakenet.json#L43), [second](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/common/coininfo.py#L1567)).

Thanks.

AlexITC, 2020-08-20:
There was no answer about this, we are interested to know:
- Would you be willing you merge this 2 lines change?
- If you are, in which release date do you think it can be included?

Thanks.

prusnak, 2020-08-20:
If you send a PR today, the change might get released on 2nd of September.

AlexITC, 2020-08-20:
PR submitted (#1208), thanks!

closed 2020-08-20 by AlexITC