trezor/trezor-firmware #1206: Show wrong address path warning also when show_display is False in get_address

tsusanka, 2020-08-19:
In Legacy we do not show warning on Trezor if an address and path mismatch occurs and the check on display is not required.

For `trezorctl get-address -c Bcash -t address -n "m/44'/0'/0'/0/0" -d` you get the warning but for `trezorctl get-address -c Bcash -t address -n "m/44'/0'/0'/0/0"` you do not (the `-d` is mising).

I think the warning should still be present, shouldn't it? Because the host requires some unusual action.

tsusanka, 2020-08-31:
We have concluded this does not pose any risks since nothing is displayed on Trezor. Let's investigate if this happens on Core as well and then considering closing this.

tsusanka, 2020-09-14:
As agreed: this is correct behavior. In Core this gets a bit more complicated as due to the Keychain. We need to skip the check for cases when the `-d` is missing. That way the check dialog won't be displayed as well and it will be in sync with T1. The only difference is that in Core this will lead to an error (because of the Keychain) which is the correct behavior anyway.

So tl;dr: we need to skip the dialog when `-d` is missing and that's it.

closed 2020-11-05 by matejcik

bosomt, 2020-12-21:
QA OK

Nixos
Suite be4e3a36f7dbe15b2b203f66de11edc50a6665b8
Device: fw-regular-2.3.5-889a3699

commit c8f10b2f (2020-10-26): feat(core/bitcoin): do not show path warning when GetAddress is called silently (fixes #1206)
commit 13a7d5e6 (2020-10-26): feat(core/bitcoin): do not show path warning when GetAddress is called silently (fixes #1206)
commit 5b2b0d8a (2020-10-26): feat(core/bitcoin): do not show path warning when GetAddress is called silently (fixes #1206)
commit 12f0f758 (2020-11-05): feat(core/bitcoin): do not show path warning when GetAddress is called silently (fixes #1206)