trezor/trezor-firmware #1207: Initializing base classes

andrewkozlik, 2020-08-20:
As someone who is coming from C++ I would like to see us calling `super().__init__()` every time we define an initializer in a derived class to ensure proper initialization of the parent's variables. See https://github.com/trezor/trezor-firmware/pull/1180 for a concrete example where this would make sense in my opinion. To me it seems counterintuitive not to initialize the parent by default, but maybe the pythonic way of thinking about this is different? Let's discuss IRL.

tsusanka, 2020-08-31:
Let's see if flake or similar can check this. Related to #132.

mmilata, 2020-11-20:
pylint [can check this](https://pylint.readthedocs.io/en/latest/technical_reference/features.html#classes-checker-messages):

> super-init-not-called (W0231):
>    \_\_init__ method from base class %r is not called
>    Used when an ancestor class method has an \_\_init__ method which is not called by a derived class.

PR #1364 (2020-11-30): Call base class __init__ in subclasses
Fixes #1207. I tried to copy the pylint's "super-init-not-called (W0231)" approach - child class must call the ancestor's `__init__` provided it has one. Number of allocations across device tests increases by about 0.5 %.

* The changes affect the generated protobuf classes, not sure if there can be impact on performance other than the small amount of allocations.
* Children of `Hash143` do a bit of unnecessary work now, perhaps it would be better to refactor `Hash143.__init__` or add an abstract base class ...
closed 2020-11-30 by mmilata