trezor/trezor-firmware #1222: Firmware error message in Wallet after setting a white noise screen

sorooris, 2020-08-24:
**Describe the bug**
A Firmware error message is displayed after setting a white noise homescreen in MyTrezor Wallet.

**Firmware version and revision**
2.3.3 e137f19

**Setup:**
 - Wallet software: MyTrezor production [commit](https://github.com/satoshilabs/mytrezor/commits/ec6b4d36cbae2ec1c3f479eb3c1891090790b7c0)
 - OS: any (tested NixOS)
 - Browser: any (tested Chromium 84.0.4147.89)

**Steps to Reproduce**
1. Navigate to _Homescreen settings_ in Wallet,
2. Select and upload a white noise homescreen,
3. Confirm and see error.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Console returns `trezor.js] [call] Received Failure`.

**Screenshots**
[image]

sorooris, 2020-08-24:
Same behavior in Suite: beta, [commit](https://github.com/trezor/trezor-suite/commit/74b74b0c70658ff4cad66541441aa7274f949308)
2.3.3 e137f19

> [image]

tsusanka, 2020-08-24:
@sorooris could you append the white noise image here pls? I think we should write a test for it :)

sorooris, 2020-08-24:
@tsusanka 
[image]

tsusanka, 2020-08-25:
I have added test in b2c63b72d6dea1aa87e62fd47776a1aaddb2e97a and it indeed fails even on an emulator. @mmilata could you look into it?

```
File "trezor/loop.py", line 191, in _step
  File "apps/base.py", line 242, in wrapper
  File "apps/management/apply_settings.py", line 52, in apply_settings
  File "apps/management/apply_settings.py", line 27, in validate_homescreen
AttributeError: 'Display' object has no attribute 'toif_info'
```

matejcik, 2020-08-25:
isn't this a duplicate of #1114 ?
can you reproduce the problem using trezorctl on a freshly plugged in TT? If not, it's a duplicate.

do other homescreens work? if yes, @tsusanka's problem is something else (mismatched trezor-core build perhaps?)

matejcik, 2020-08-25:
also let's not add whitenoise to tests, or else mark them as xfail or something; whitenoise homescreen is currently not guaranteed to work on hardware unless it's freshly plugged in

tsusanka, 2020-08-25:
Ah, I saw `File "apps/management/apply_settings.py", line 27, in validate_homescreen` and directly thought it was introduced by https://github.com/trezor/trezor-firmware/issues/1118. Ok @sorooris please retest what @matejcik says.

sorooris, 2020-08-25:
@matejcik please could you specify what "freshly plugged in TT" is?

Using the same white noise screen file with trezorctl (`trezorctl set-homescreen -f white-noise.png`) with a reconnected 2.3.3 e137f19 returns the following:

`[seb:/home/dir]$ trezorctl set-homescreen -f white-noise.png 
Traceback (most recent call last):
  File "/nix/store/pacl0pp9jnd6p706qw9a8mxkxqpx3ar8-python3.7-trezor-0.12.0/bin/.trezorctl-wrapped", line 9, in <module>
    sys.exit(cli())
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/nix/store/cqxkmww7dlic0qfskpmcd3xmckhsf6mi-python3.7-click-7.1.2/lib/python3.7/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/nix/store/pacl0pp9jnd6p706qw9a8mxkxqpx3ar8-python3.7-trezor-0.12.0/lib/python3.7/site-packages/trezorlib/cli/__init__.py", line 78, in trezorctl_command_with_client
    return func(client, *args, **kwargs)
  File "/nix/store/pacl0pp9jnd6p706qw9a8mxkxqpx3ar8-python3.7-trezor-0.12.0/lib/python3.7/site-packages/trezorlib/cli/settings.py", line 114, in homescreen
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'
`

tsusanka, 2020-08-25:
@sorooris are you in our repository inside `nix-shell`? Because this sounds like a "host" problem

matejcik, 2020-08-25:
@sorooris "freshly plugged" is you unplug, then replug, then do the thing without doing anything else with the device.

please also try setting the whitenoise homescreen as first thing through wallet. @bosomt was successful with that last time

prusnak, 2020-08-25:
@seb `No module named 'PIL' ` => your installation is missing python-pillow package; i guess we need to fix that in the nix package

@tsusanka `AttributeError: 'Display' object has no attribute 'toif_info'` => it seems you forgot to rebuild the emulator

matejcik, 2020-08-25:
Ah, also trezorctl doesn't support png files for TT.

we have code that enables it, but it's currently internal ... i'm a little reluctant to move it to the main package, but perhaps we could?

tsusanka, 2020-08-25:
Confirming the test I have written has passed after I have rebuilt _core_. Apologies :roll_eyes:.

sorooris, 2020-08-25:
@matejcik Thanks for clearing that up:
using the _bad-noise.toif_ file with `trezorctl set-homescreen -f bad-noise.toif` command works and returns the correct error [message](https://github.com/trezor/trezor-firmware/issues/1118#issuecomment-678980669). However, I'm unable to verify in Wallet/Suite since .toif files are not allowed there.

matejcik, 2020-08-25:
please try with "good" noise.toif too. If that succeeds from trezorctl on a freshly-plugged TT, we can close the issue as a duplicate of #1114

sorooris, 2020-08-25:
QA OK
The "good" white noise (white-noise.toif from Slack) is set seamlessly via trezorctl; let's close it as a duplicate of the #1114 issue.

matejcik, 2020-08-25:
thanks! closing

closed 2020-08-25 by matejcik