trezor/trezor-firmware #1224: Pillow missing from requirements.txt

prusnak, 2020-08-25:
python-trezor imports `PIL` in several files (e.g. `python/src/trezorlib/cli/settings.py`, `python/src/trezorlib/_internal/toif.py`) but the dependency `Pillow` is missing from `requirements.txt`

or should it rather belong to `requirements-optional.txt` ?

matejcik, 2020-08-25:
it's optional so it should be in requirements-optional.txt
but then there is no straightforward way for users to install it.

if it isn't present in the pipenv or nixshell, that should probably be solved there, right?

prusnak, 2020-08-25:
> if it isn't present in the pipenv or nixshell, that should probably be solved there, right?

Right. It is already present in Pipfile/pyproject.toml, so this was just a formality (so packagers for other distributions looking into `requirements.txt` do not miss this dependency.

commit eddaeb12 (2020-08-25): python: add Pillow as an optional requirement (fixes #1224)
closed 2020-08-25 by matejcik