trezor/trezor-firmware #1226: Nano Support

My1, 2020-08-26:
**Describe the solution you'd like**
Support the Cryptocurrency nano

**Describe alternatives you've considered**
The Ledger devices support nano but are not open Source. The Keepkey (which forked T1, so maybe you could even get some code from there) also supports Nano but only on the firmware wallet with no companion currently available to support it due to missing library support.

**Additional context**
Nano tries to be true to the spirit of cryptocurrencies, actively encouraging decentralization, having fast and fee-less transactions without relying on a second Layer implementation that has its own problems, or overly complicated stuff like smart contract, just plain and simple.

Also, with Exodus there would be even already a companion Wallet that can support nano.

Needed Algorithms are Blake2b for hashing and Ed25519 for Signing.

BIP39 Data.

Derivation Path: 44'/165'/x' (x is the nth account starting from 0)
also quote from the Nano Docs:
https://docs.nano.org/integration-guides/key-management/
> The BIP39 seed modifier "ed25519 seed" is used which makes wallets compatible with each other

not exactly sure what it means but probably important.

matejcik, 2020-08-26:
We are currently only adding coins that are in the Top 30 by market cap, Nano is at n. 77.
If it gets to Top 30, and someone wants to contribute a PR implementing the support, we'll be happy to consider it for inclusion.

closed 2020-08-26 by matejcik

My1, 2020-08-26:
frankly kinda sad especially considering the trezor supports already almost 1200 assets already.