trezor/trezor-firmware #1231: Should Cardano usage emit a new session id?

tsusanka, 2020-08-27:
An example from @szymonlesisz:

1. I am using BTC:
```
-> Initialze(session_id=None)
<- PassphraseReq()
-> PassphraseAck(passphrase="A")
<- Features(session_id="ABCD")
```

2. I switch to Cardano:
```
-> Initialze(session_id="ABCD")  # I use the very same session_id received previously
<- PassphraseReq()  # passphrase is requested again though because of Cardano
-> PassphraseAck(passphrase="B")  # I enter different one
<- Features(session_id="ABCD")  # I get the same session id
```

@szymonlesisz was quite surprised the session id has not changed because under the `session_id="ABCD"` we currently have two passphrases (one for BTC, one for Cardano).

Should we:
1. Create a new session anytime Cardano is used for the first time. In another words we would need to track whether the session is "normal" one or "Cardano" one.
2. Simply derive the Cardano seed anytime passphrase is being exchanged for other coins. This is ugly but it will simplify so many things and is properly a requirement for Suite implementation at some point anyway.
3. Not deal with it at the moment because there is not (hopefully) a mixed wallet yet anyway.

cc @matejcik

matejcik, 2020-08-27:
we could go for a hybrid of (1) and (2):
- whenever Cardano key is requested, and the session is clean, derive both the normal seed and the Cardano seed
- if the normal seed is already derived but Cardano seed isn't, raise "invalid session", let the caller `Initialize` again and request a Cardano key as the first thing.
Suite can use `CardanoGetPublicKey` upon initialization to deliberately trigger this behavior.

advantage: we save time and space for Cardano seed derivation for the majority usecase
disadvantage: the behavior is rather weird.

i'd go with (3) for now, but I'm afraid we will end up doing (2) for practical reasons, though it's nasty.
Couldn't we throw out the Cardano derivation instead? :)

if we were concerned about derivation time, I suppose we could store some sort of intermediate result for Cardano, which would need to get processed before using as a seed; but I'm not sure if that's secure or if it brings any advantage.

tsusanka, 2021-06-18:
@vladimirvolek @mmahut this is a prerequisite for Cardano in Suite and we'll have to require a FW which has this change before letting them use Cardano in Suite.

---

I actually like @matejcik's model but we need to make sure how Connect responds to the `Invalid session`. @szymonlesisz is Connect handling that?

tsusanka, 2021-06-18:
@matejcik or should we first measure option (2)? We could draft it and simply push to CI on our HW tests and compare the times with current setup?

matejcik, 2021-06-18:
I don't think we have any other options than some variant of (2). Simple reason: anything else you do, you are asking the user for passphrase twice.

I believe the Suite-side requirement is only one passphrase prompt, which also implies that it's impossible to use different passphrases for Cardano and rest of coins?

matejcik, 2021-06-18:
also ISTM when we have Cardano in Suite, we stop caring about performance for non-Cardano non-Suite users? :)))

An unpleasant feature of Cardano key is that it's relatively big - in addition to 64 bytes per session we have currently, we'll need additional 96 bytes per session for the derived Cardano key. We reserve space for 10 sessions, so that's 960 more bytes in total. I don't see any possible way of getting around that though, except tell the users to upgrade to Shamir backup....

prusnak, 2021-06-18:
> We reserve space for 10 sessions, so that's 960 more bytes in total.

10 sessions seems too much - we might decrease this to e.g. 4-5 sessions

andrewkozlik, 2021-08-18:
Firstly, I think our objective should be to make it as simple as possible for the user, i.e. enter passphrase only once, because having to enter the passphrase multiple times discourages them from using a secure passphrase. So there are these possibilities:

1. Cache the passphrase and derive the Cardano seed only if and when needed.
    - Advantage: Saves derivation time.
    - Disadvantage: Caching the passphrase is not best practice.
    - Opinion: Acceptable.
2. Always derive both the standard seed and the Cardano seed.
    - Advantage: Doesn't cache the passpharse in cleartext.
    - Disadvantage: Cardano derivation takes 4096 iterations of HMAC-SHA512 whereas standard derivation takes 2048 iterations. Furthermore Cardano needs to derive two blocks, so altogether it takes four times as long. This means that the total derivation time will be quintupled. The result that will be cached is 96 bytes in size.
    - Opinion: We'd need to measure the time, could be 5 sec maybe.
3. Store some sort of intermediate result for Cardano and compute the seed only if and when needed.
    - Advantage: Saves derivation time. Doesn't cache the passpharse in cleartext.
    - Disadvantages: If we start off the passphrase hashing, the intermediate result is 128 bytes in size. To complete the PBKDF2 computation either the mnemonic is required or another 128 bytes of intermediate data. We would have to stick our fingers into the SHA implementation and the PBKDF2 implementation.
    - Opinion: Seems complicated.

> we'll need additional 96 bytes per session for the derived Cardano key

I don't see a way around this, other than caching the passphrase and rederiving the seed every time.

matejcik, 2021-08-19:
A thing to note that I realized yesterday:
If we cache the passphrase and not the derived Cardano seed, we need it to re-derive it _for every Cardano operation_, in particular also for every `GetPublicKey`, `GetAddress`. This might be significant for Suite usage.

andrewkozlik, 2021-08-19:
> A thing to note that I realized yesterday:
> If we cache the passphrase and not the derived Cardano seed, we need it to re-derive it _for every Cardano operation_, in particular also for every `GetPublicKey`, `GetAddress`. This might be significant for Suite usage.

It seems to me that that's what it's doing now. The only difference is that the current implementation asks for standard passphrase and Cardano passphrase separately.

Anyway, what I had in mind (but didn't make clear) for Solution 1 was that once the Cardano seed is derived, it would be cached. To save space there could be a union-like structure in the cache containing either the passphrase or the derived Cardano seed and the first byte would indicate which one it is. But I suppose that doesn't fit very nicely into the current decorator implementation.

matejcik, 2021-08-19:
If we reserve the space for Cardano seed, it just seems nicer and overall simpler to derive it every time alongside the normal seed. The only problem is that it adds some upfront cost per session initialization.

ISTM we need to measure how big that cost actually is.

onvej-sl, 2021-08-19:
This is how much time it takes to derive a seed from a mnemonic and a passphrase on TT. The times for T1 will be about 40 % higher.
|                         | 128-bit mnemonic | 256-bit mnemonic |
|-------------------------|------------------|------------------|
| bip39 seed derivation   | 580 ms           | 580 ms           |
| slip39 seed derivation  | 800 ms           | 840 ms           |
| cardano seed derivation (Icarus) | 2330 ms          | 2340 ms          |

The length of the passphrase has a negligible effect on the speed. The cardano seed derivation (compared to the bip39 and slip39 derivations) [includes](https://github.com/trezor/trezor-firmware/blob/0c1b56a3b81688992d95493d6095b1bc6370c1e6/crypto/bip32.c#L416) also a derivation of a [public key](https://github.com/trezor/trezor-firmware/blob/0c1b56a3b81688992d95493d6095b1bc6370c1e6/crypto/bip32.c#L380), which takes about 10 ms. 

See [this](https://github.com/trezor/trezor-firmware/blob/onvej-sl/cardano_measure/core/src/measurement.py) for details.

tsusanka, 2021-08-20:
A summary from the product side as discussed with @Hermez-cz:
1. Worsening experience for non-Cardano users by 2 seconds is a no-go.
2. Solving this somehow in Suite UI (e.g. requesting passphrase again for Cardano users) is a no-go as well as this will introduce problems elsewhere mainly in the Suite Dashboard.
3. Caching passphrase in plaintext is mostly our decision but Product does not like it much as we usually claim "passphrase stays in Trezor for as short period of time as possible".

----

I think the next step is to write up our ideas about how we could improve the protocol to introduce "Cardano on-demand derivation".

matejcik, 2021-08-20:
one option is to keep a flag indicating whether Cardano seed should be derived or not

variant A: specify at session initialization time, applies to current session
* drawback: if the user enables Cardano in a running session, they will need to re-enter passphrase for all their hidden wallets
  - this happens once per Suite installation (so again on another PC...)

variant B: store and remember flag in flash, applies to all sessions until unset.
* drawback: if you try Cardano once and then never again, the flag still makes your Trezor slower
* drawback: this also holds for non-Cardano wallets (metamask...)
* same drawback as in variant A, except it only happens once per lifetime of Trezor

andrewkozlik, 2021-08-25:
Partially related: https://forum.cardano.org/t/swapping-from-ledger-to-trezor/50513/7. The incompatibility with Ledger and incompatibility with Icarus when the mnemonic is 24 words is another problem which deserves consideration as we think about how to solve this.

matejcik, 2021-08-31:
Aside: we currently can, and should, check that Cardano passphrase is identical to the main passphrase.

If Cardano passphrase is asked first, this is trivial: derive the normal seed as well.

If it's asked second, try using the passphrase to re-derive the normal seed. If it comes out different, reject passphrase.

Not sure if this is worth implementing separately from the proper fix as discussed here.

matejcik, 2021-10-08:
@onvej-sl do your results take into consideration the `USE_BIP39_CACHE` feature?
The result of mnemonic derivation for BIP39 seems to be cached on the C level in practice.

closed 2021-11-10 by matejcik

matejcik, 2021-11-10:
QA: this will only work after the matching Connect PR is deployed: trezor/connect#933
Check that Adalite and Yoroi are functional, you can generate a receiving address and display it on Trezor screen, both with Single and Shamir backup

PR #1873 (2021-11-10): Cardano derivation improvements
With this PR, the same passphrase is used for Cardano derivation and the standard derivation (fixes #1231). If enabled via `Initialize(derive_cardano=True)`, the Cardano derivation runs at the same time as the standard derivation, which means that the start-up time is longer.

This is only relevant for BIP-39 seeds. With SLIP-39, a different derivation is used and it is available even without `derive_cardano=True`.

Additionally, the Ledger and Icarus derivations are implemented. Ledger derivation is available even in sessions that don't have `derive_cardano=Tru [... cut]