trezor/trezor-firmware #1232: python-trezor: missing attr from requirements.txt/setup.py

prusnak, 2020-08-27:
```
ModuleNotFoundError: No module named 'attr'
```

can we add attr from `requirements.txt`/`setup.py`?

Even better - can we have some check for this?

prusnak, 2020-08-27:
It seems there is a flake plugin for this: https://pypi.org/project/flake8-requirements/

matejcik, 2020-08-28:
this particular thing should have been found by `tox`, i'll need to investigate why that didn't happen

matejcik, 2020-08-28:
ok, this is fun: `attrs` is a dependency of several things in the dev environment, and tox is testing inside that.

pradysp, 2020-08-29:
Hello is there a workaround for this bug ? I cant even run the basic '**trezorctl**' command now.

prusnak, 2020-08-29:
@pradysp which distro are you using?

pradysp, 2020-08-29:
@prusnak Using Anaconda distribution of python version 3.6, trezor 0.12.2 release

prusnak, 2020-08-29:
@pradysp install the `attrs` package into your environment

pradysp, 2020-08-29:
@

> @pradysp install the `attrs` package into your environment: most probably `pip install attrs`

That fixed it. Thanks

matejcik, 2020-09-29:
fixed and added a check in 6dda240f

closed 2020-09-29 by matejcik

AdamSHurwitz, 2021-07-13:
The workaround with `pip3 install attrs` is still required in *version 0.12.2* or else an error is thrown that `attrs` is not found.