trezor/trezor-firmware #1237: make protobuf enums public

matejcik, 2020-09-01:
several times now we needed access to the `protobuf.EnumType` object (usually to convert the value to and from a string), and for type hints we often use `EnumType<Something>`

trouble is, `EnumTypeSomething` is only accessible from `ProtobufTypeWhereEnumIsUsed` but not from the `Something` itself.

even worse, the `protobuf.EnumType` instance is only accessible from _the field definition_ where it is used, i.e., must be parsed out from `ProtobufTypeWhereEnumIsUsed.get_fields()`

let's figure out a way to publicize the EnumType instance

matejcik, 2020-09-03:
see also https://github.com/trezor/trezor-firmware/pull/1240/files/1ca4ff8e9967a9765b85c37bf21c830e78ade4d6..f33b10674ab48f552920b19e081a965efe011fcb#diff-c4876a922c4f5c83d42166249b74c6aaR49-R50 where there's the same enum used twice in a message

mmilata, 2020-09-29:
Can we somehow make this work as [Python's enum.Enum](https://docs.python.org/3.7/library/enum.html)?

matejcik, 2020-09-29:
on the trezorlib side, probably.
on the upy side, probably not, they don't have `Enum` AFAICT

matejcik, 2021-09-07:
this is done as of #1541

closed 2021-09-07 by matejcik