trezor/trezor-firmware #1245: Additional Cardano features

MichalPetro, 2020-09-03:
We would like to propose to add the following functionalities for Cardano model T integration:

- Review and polish how users interact with accounts. Cardano main wallets will be adding multi-account support in order to allow partial stake delegation (each account will be delegating to a single stake pool) and we would like to improve the UX for working with accounts and also review the existing security limitations related to accounts (such as display warnings for exports of keys for account 20+, etc) as this was not our main focus when implementing the Shelley features.

- Add support for Stake Pool owners (people that do not operate pools but pledge the funds to the pool). This will require a new flow where input will be a transaction with a pool registration certificate, Trezor will parse the certificate and display important data to the user, the output will be a staking key witness. (owners will send the witnesses to pool operator who will assemble the transaction with collected witnesses - not related to Trezor just so you understand why). We will try to abstract this feature so we could use it as a baseline for proper Cardano multi-sig in the future.

Additionally, stake pool operators would like to have support on HW wallets but you didn't show interest for this feature in the past. Just checking now if you will be interested in support for stake pool operators which will require management of completely new unrelated cold key (no new crypto), signing complex pool registration and less complex deregistration transactions, creating an operational key certificate from a combination of cold key and provided hot key (some time sequencing involved). Most likely CLI only.

After we receive the final specification, we would also like to add multi-sig support but no ETA for this yet.

MichalPetro, 2020-09-09:
@tsusanka Currently, we are defining multi-account support for Cardano and we discussing batch exports of account pub keys for the account discoveries. We noticed that Trezor allows a potentially unlimited amount of pub key exports just by confirming a single prompt which is asking for confirmation to "export multiple public keys" in Trezor Connect. What is the trust model that you considered when you were allowing/implementing this feature? Isn't it potentially a privacy leak to give the wallet the possibility to export multiple public keys without notifying users which keys are they exporting? 

Ref: https://github.com/trezor/connect/issues/679

tsusanka, 2020-09-15:
> Review and polish how users interact with accounts. 

Any UX improvements are extremely welcome! However, we plan to redesign the TT dialogs a bit (hopefully later this year) so some things might change. So I am not sure it makes sense to do it now? But we will still keep "pagination" etc so it shouldn't be something huge.

> Add support for Stake Pool owners 

This hugely depends on the amount of code. Any idea how much the Cardano would increase? Like 10% or twofold? :)

MichalPetro, 2020-09-21:
For now just this one PR that is fixing UI to work well with higher account indexes
https://github.com/trezor/trezor-firmware/pull/1274

Regarding the Stake Pool owners support - this should be really a very simple thing that will be most likely in a scale 3%-6% increase.

MichalPetro, 2020-09-23:
@tsusanka do you think you will be able to squeeze this tiny PR into next FW release? Also - what is the code freeze date for November release so we can plan our following work accordingly?

tsusanka, 2020-09-25:
Sorry, next release will consist only of cherry-picking the Monero HF commit. See our [changelog](https://github.com/trezor/trezor-firmware/blob/master/core/CHANGELOG.md). 

Next release will be either on Nov 4th or Dec 9th (first Wednesday of a month). If nothing important comes that needs to be released asap we will most likely go with Dec 9th.

Freeze is always ten days before the release day.

disassembler, 2020-10-07:
I'd like to see this land Nov 4th if we can. This feature is very important for a lot of stake pool operators and larger organizations to be able to fully pledge their pools as owners from hardware wallets.

MichalPetro, 2020-10-19:
@tsusanka we have the PR for stakepool owner support ready. In the end, it is a bit more code than expected but still below 15% of total volume (including some re-factors)

Additionally, I have a question? IOHK is considering introducing "hybrid wallets", i.e. wallets that would have a "cold" key for fund transfers and "hot" key for staking.

For that usecase, IOHK would like to be able to derive and export a staking private key from the hw wallet which would be usable in a hot environment that is disconnected from the hw wallet, but at the same time the hw wallet itself can do the staking if desired/needed.

Question: Is it viable to allow export of private keys from the HW wallet for a certain HD path or do you have a policy that strictly forbids private key exports under any circumstances?

tsusanka, 2020-11-04:
@MichalPetro do I understand it correctly you would like to export smth like `m/44'/X'/0'` both its private and public key?

MichalPetro, 2020-11-13:
> @MichalPetro do I understand it correctly you would like to export smth like `m/44'/X'/0'` both its private and public key?

The path would be very specific, more like m/1852'/1815'/0'/2/X

MichalPetro, 2020-11-26:
@tsusanka I received a word that you are going to pass the December release, is that true? We were somehow relying on this in Cardano community and some network change plans are closely tight to the features delivered by this update. Currently, the stake pool operators are unable to safely pledge funds to their pools which is damaging the ecosystem. Could you guys reconsider, please? Ref: https://github.com/trezor/trezor-firmware/pull/1317

tsusanka, 2020-11-27:
Yes, it is. We have not finished some features we wanted to release in Dec 9 and other than that there are no significant features for Suite/Wallet users. I know I have mentioned Dec 9 so I apologize for not fulfilling on that promise but we simply can't do releases based on other coins' plans. We are already getting a lot of feedback that the updates are way too often and we are trying to minimize that. We can't push an update that does not bring anything to our Suite users. Thanks for your understanding

MichalPetro, 2020-11-27:
The upcoming increase of network k-parameter (desired number of stake pools) results in decreased stake pool saturation cap and the community will need to divide their funds into more stake pools. Trezors' owners are users with large ADA holdings and the only most secure way for them to handle delegation to multiple stake pools is by having the multi-accounts and pledge features available within this firmware update. Particularly the cold key pledge feature is really demanded by the community and it will help stake pool operators and owners to protect their funds. I understand your stance and your primary focus on BTC and suite users, but by omitting this release, you are exposing the Cardano Trezor users with large holdings to unnecessary risk. Is it possible to release some non-mandatory FW update or this is completely out of question?

AlexandrJerechinsky, 2021-01-11:
The firmware release containing the multi-accounts and pledge features was expected to come out this January, but unfortunately had to be postponed until February due to time restrains of other stakeholders. We apologise very much, as we are aware of the risks Cardano Trezor users are facing due to lack of these features. We plan on including #1390 and #1396 in the February release as well.

MichalPetro, 2021-01-19:
@AlexandrJerechinsky @tsusanka 

IOHK informed us that at the end of February there will be a hard fork (Mary) that will extend the tx output format to add support for multiple assets, breaking compatibility with wallets that do not update as they won't be able to spend from such utxos (that contain ADA as well). We expect to have this update for Trezor ready for review by the end of next week - could you please include that into your release plan as well, accounting for the fact that it may be ready close to or during the pre-release code freeze?

refi93, 2021-01-20:
@prusnak @AlexandJerechinsky @tsusanka - We noticed that with Ethereum you explicitly forbid operations with custom tokens, only allowing the (statically) whitelisted ones: https://github.com/trezor/trezor-firmware/issues/798 - given the way tokens are implemented in Cardano this however is not possible given that ADA and arbitrary tokens can be coupled together in a single utxo and the user does not have control over which tokens can arrive in such utxo alongside the whitelisted tokens/ADA, even if there was such whitelist and they would still need to explicitly move them to the tx output in order for the transaction to be balanced. Therefore we would like to here your opinion/advice on the subject before we include that functionality.

The Mary hardfork introduces the concept of custom native tokens in the Cardano blockchain, making them part of the utxos which also contain ADA, the native asset. This means that in order to prevent the ADA in such utxos from being locked, Trezor needs to support transfering such assets and including them in the transaction outputs, as the inputs and outputs need to be balanced also in terms of the tokens. Tokens in Cardano will be uniquely identified by their policy id (28 byte hash) and asset name (an ASCII string up to 32 characters long). There can be multiple differently named assets within the same policy.

Here are Cardano developer docs for more context: https://developers.cardano.org/en/development-environments/native-tokens/working-with-multi-asset-tokens/

> Note: The transaction has to balance: the amount of lovelace in the output (plus the fee) has to be the same as the value in the consumed UTxO entry; and the amount of tokens in the output has to match the value that is specified in the --mint field.

and Cardano's CDDL for Mary era defining the new structure of tx outputs: https://github.com/input-output-hk/cardano-ledger-specs/blob/d99dcda8c06a1bad301ebd4cad56cf51ab03852d/shelley-ma/shelley-ma-test/cddl-files/shelley-ma.cddl#L250

Given the general usability concern, we intend to support sending arbitrary tokens in Cardano, showing to the user explicitly the policy id and asset name which is up to them to check similarly as they would check the output address - is that a viable approach for you?

matejcik, 2021-01-20:
> you explicitly forbid operations with custom tokens, only allowing the (statically) whitelisted ones

this was never the case, and it's even more untrue today with #800 being fixed :) we always supported sending any token, but in case of Ethereum, we could not display it properly because we didn't know the denomination and currency symbol.

The whole issue with tokens is the problem of unique identification: in the presence of a valuable XYZ token and a cheap token also called XYZ, it should be difficult to mislead the user as to which of the two they are transacting.

If Cardano signs over both the policy id and the asset name, so this is data that doesn't need to be trusted, it is not a problem. We similarly allow arbitrary tokens on Binance Chain and EOS.

refi93, 2021-01-20:
@matejcik ok great - sorry for the confusion and thanks for clarifying :)

tsusanka, 2021-01-20:
> IOHK informed us that at the end of February there will be a hard fork (Mary) that will extend the tx output format to add support for multiple assets, breaking compatibility with wallets that do [...]

@MichalPetro we will _freeze_ this Friday and finalize and sign on March 3rd. So I am afraid we won't make it. However we will try to talk to Cardano foundation that they really really need to inform us about these changes months beforehand not weeks or days.

That being said, feel free to prepare the PR and let's discuss then.

Lvceo, 2021-04-21:
No ADA support for Trezor One in the roadmap?

matejcik, 2021-04-27:
i'm wondering if we can close this? many Cardano PRs have landed in the meantime
ping @MichalPetro

MichalPetro, 2021-04-27:
yes I am closing this, thank you guys for your cooperation.

closed 2021-04-27 by MichalPetro