trezor/trezor-firmware #1251: Error building/compiling Trezor One bootloader

vahidtaghiloo, 2020-09-05:
I was trying to compile Trezor one bootloader and firmware on Debian Linux . The firmware was built successfully but building bootloader throws an error like this: "ERROR: Couldn't install package: markupsafe"
I recorded the terminal with asciinema and uploaded here: https://asciinema.org/a/357819
I don't know what I'm doing wrong :)

**Firmware version and revision**
legacy/blv1.8.0

prusnak, 2020-09-05:
You can build an old tag using the new build script. Try checking out the master branch and run `./build-docker.sh legacy/bl1.8.0`

I am closing this, because this can't be fixed in an old tag anyway. But feel free to continue the discussion here.

closed 2020-09-05 by prusnak

vahidtaghiloo, 2020-09-05:
Thank you for your time. Unfortunately, building from the master branch gives exactly the same error. I wonder if something is wrong with my OS. Is there a way to download a precompiled bootloader binary for Trezor One?

https://asciinema.org/a/357988

prusnak, 2020-09-05:
>  Is there a way to download a precompiled bootloader binary for Trezor One?

Here: https://github.com/trezor/trezor-firmware/blob/master/legacy/firmware/bootloader.dat

vahidtaghiloo, 2020-09-06:
Thanks a lot. Any idea how to flash bootloader.dat file on microcontroller? Or how to convert .dat file to .bin/.hex?
I tried to convert it with xdd but it doesn't seem right.

prusnak, 2020-09-06:
The file is a .bin file, just flash it at start of the flash.

thohemp, 2020-11-13:
I can report the same error.