trezor/trezor-firmware #1256: mismatched fingerprints for core/v2.3.3

syzygyzer, 2020-09-12:
**Describe the bug**
Possibly related to https://github.com/trezor/trezor-firmware/issues/692

I am attempting to build the latest Trezor Model T firmware following the instructions at https://wiki.trezor.io/Developers_guide:Deterministic_firmware_build, except I checkout core/2.3.3 instead of core/2.3.0 it currently states.

I am running a fairly fresh minimal install of Debian Stable 10. I have userspace managed mostly with Nix, but this is the first Python build I've done on this system so far and installed the prerequisites up through `wheel` listed here globally https://wiki.trezor.io/Using_trezorctl_commands_with_Trezor using pip3. But theoretically my system's environment shouldn't matter inside of a Docker -> Nix -> `pipenv` triple combo! This is also the first build I've done with `docker` on this system, so all I have installed is the most recent stable version.

I've tried the build a few with minor variations a few times and the output fingerprints have always been the same. I followed it as written the first time. I noticed a message saying `LANG` was unset and the compiler suggested to set it in `.profile`, so I did even though `echo $LANG` gave `en.US-UTF8`  in the shell (for which I tried both `fish` and `bash`). Each time I got some intermediate fingerprint messages that look like this:

```
Vendor Header for SatoshiLabs version 0.0 (4608 bytes)                                                                                                                                                                                       
✔ Signature is VALID                                                                                                                                                                                                                         
Firmware Header {                                                                                                                                                                                                                            
    magic: HeaderType.FIRMWARE                                                                                                                                                                                                               
    header_len: 1024                                                                                                                                                                                                                         
    expiry: 0                                                                                                                                                                                                                                
    code_length: 1079296                                                                                                                                                                                                                     
    version: 2.3.3 build 0                                                                                                                                                                                                                   
    fix_version: 2.2.0 build 0                                                                                                                                                                                                               
    hashes: [                                                                                                                                                                                                                                
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ❌ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                  
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
        ✔ 0000000000000000000000000000000000000000000000000000000000000000                                                                                                                                                                   
    ]                                                                                                                                                                                                                                        
    sigmask: 0                                                                                                                                                                                                                               
    signature: 64 bytes 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                     
}                                                                                                                                                                                                                                            
[... cut ...] MISSING, hashes are MISSING` seems odd. I have never tried to build the Trezor software before today though.

I also tried to build core/v2.3.0 once but it failed to complete somehow - not sure if this will help diagnose my issue with v2.3.3. I have not tried to build any other versions yet.

**Firmware version and revision**
core/v2.3.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: trezor-firmware
 - OS: Debian Stable 
 - Version - 10

Thank you!

prusnak, 2020-09-12:
We should make sure the hashes are added to the firmware header first and compute the fingerprint only after this has been done. Can you please have a look at this @mmilata? Maybe @matejcik can help with the tooling.

matejcik, 2020-09-14:
how are the final fingerprints calculated? fingerprint for core should be calculated by `headertool.py --digest`, which takes care of the missing hashes internally. Or alternately `headertool.py --rehash` should be used to insert the hashes; this usually happens in the same step as signing, as I thought that hashing without signing is kind of pointless.

matejcik, 2020-09-14:
oooh i see, it's using the `header_digest` function from `trezorlib.firmware`
well, that's not right.

IMHO the behavior of `header_digest` is correct in ignoring the bad hash values; there is no signing capability and if you tried to upload hashless firmware, it would in fact have the digest that `header_digest` claims.

So the solution is to either use `headertool` for generating the `.fingerprint` file, or using `headertool --rehash` before

prusnak, 2020-09-14:
@matejcik it seems martin used the `trezorlib.firmware.digest` function because it supports both T1 and TT. I think we can use the `firmware-fingerprint.py` tool as is for T1, but use `headertool -d` for TT?

matejcik, 2020-09-14:
yes, that's reasonable

mmilata, 2020-09-14:
Thank you for quality bug report, @syzygyzer! Proposed fix is in #1261. 

Regarding the `tail -c ... | sha256sum` method from the Trezor Wiki, that is for Trezor One only. Trezor T currently has 5632 bytes long vendor header + firmware header, so the following commands should produce identical hashes:

```
tail -c +5633 trezor-2.3.3.bin | sha256sum
tail -c +5633 trezor-2.3.3-your-build.bin | sha256sum
```
`tail -c +N` seems to be 1-based and signature is at the very end, thus 5633.

PR #1261 (2020-09-15): build-docker.sh: fix fingerprint calculation for model T production builds
Fixes #1256. 

For core the fingerprints now match https://wallet.trezor.io/data/firmware/2/releases.json:
```
$ ./build-docker.sh core/v2.3.3
...
$ python/tools/firmware-fingerprint.py build/core/firmware/firmware.bin
46326222f8afcb82e1cd07867bc3bf8836f4e9d0f367e23b58d1e9bc32cd032e

$ python/tools/firmware-fingerprint.py build/core-bitcoinonly/firmware/firmware.bin
dda77cd7893a5f413f8fc4b2f44d1d43ed4b26e8ced5e6e578cc6b302c1a2310
```

For legacy the fingerprints still do not match as documented on https://wiki.trezor.io [... cut]
closed 2020-09-15 by prusnak