trezor/trezor-firmware #1259: Add performance test against master

tsusanka, 2020-09-14:
Follow-up of #1230. We currently have UI tests which turned out quite useful. We would like to introduce a very similar concept regarding performance. We could measure how long the test takes, put into some json file and then check if the time is not higher. Some sort of regression testing.

prusnak, 2020-09-14:
I got the following idea from [Casa performance testing blogpost](https://blog.keys.casa/bitcoin-multisig-hardware-signing-performance/) - maybe we can split the measurements to measure:
1) how long does it take to stream data to/from Trezor
2) how long does Trezor spend processing them

I think it might be possible to add some instrumentation to `trezorlib` to distinguish between the two.

tsusanka, 2020-09-29:
Let's start by having the 100 inputs, 100 outputs test against emulator and measure that it did not go over some threshold.