trezor/trezor-firmware #1260: Disable prevtx streaming on safety_checks=Prompt

tsusanka, 2020-09-14:
Not streaming previous transaction is dangerous as we know. However, when dealt with larger number of UTXOs the signing takes really long time. We might replace it with a warning when the safety_checks are set to Prompt.

matejcik, 2020-09-14:
what we already discussed:

* we could simply not request previous txes when safety_checks are set to Prompt. This is not ideal; at the very least, it doesn't do what "Prompt" claims to do.
* we could _prompt_ the user before every signing. That's also not ideal, it's a different kind of prompt and it's asking the user for a decision that heavily depends on the host software.

The most promising avenue: we could allow the host to indicate that it wants to skip prevtx streaming. In Strict mode this would fail; in Prompt mode the user could allow it before signing (or during the signing confirmation flow).

This seems to be doable in two basic ways:

#### (a) Add a field to `SignTx` for this.

This seems the better option.
If we did this right at start, we could have implemented the Ledger option: Host software which does not know about this option can't set it, the default value is "try to skip", the user can confirm whether they want to do that.

That ship has sailed, but it might still be worth it.

The behavior would be as follows:

add a field: `optional bool skip_prevtx_streaming;`

In Strict security, `skip_prevtx_streaming=true` would be rejected.
False or unset would mean that Trezor always asks for previous txes. This is the current behavior.

In Prompt security, `true` or unset would ask the user: "Bla bla, skip transaction verification?"
If the user says "yes", Trezor would not ask for previous txes.
If the user says "no" _and_ the field was unset, or if the field was originally set to `false`, Trezor would ask for previous txes. (this ensures compatibility with host software that does have previous txes, but does not know about this field)
If the user says "no", and the field was `true`, Trezor fails immediately.

To make the three options more explicit, the field could be an enum:
```protobuf
enum SkipPrevTxStreaming {
    TRY = 0;
    YES = 1;
	NO = 2;
}

optional SkipPrevTxStreaming skip_prev_tx_streaming [default=TRY];
```

The default of "try" even indicates to host software what it should do: have the prevtxes ready (in default configuration Trezor will fail without them), but let the power-user make the signing faster.

**Alternately**, there are only two options, either `skip_prev_tx_streaming=true` or it isn't, in which case Trezor behavior reverts to the current one.

#### (b) Allow the host to send empty previous metadata

If the host does not have the prevtx data, it can send `TxAck(tx=None)`. Upon encountering this, Trezor would display the confirmation dialog. If the user says "yes", no further previous metadata is requested; otherwise, signing fails.

I don't see any practical reason to go with (b). It is a little weird and allows for strange corner cases - can the host provide some prevtxes but not all?

tsusanka, 2021-11-18:
Do we still want to do this as this is being solved by Taproot? @matejcik @prusnak @andrewkozlik

matejcik, 2021-11-18:
From my POV, the solution I proposed is neat :) The main question is, are there any actual usecases?

prusnak, 2021-11-18:
I am not sure it is worth implementing now that the dust settled in.

tsusanka, 2021-11-22:
I am closing as Taproot solves this. Let's reevaluate if this pops up again.

closed 2021-11-22 by tsusanka