trezor/trezor-firmware #1262: Add support for TRON(TRX) alternative project

Starsakary, 2020-09-15:
**Please note this is a different project from #259. This new project from TRON official dev team. We're going to create a new project and a pull request.**

Hi. We are from TRON. TRON is a robust blockchain ecosystem with multiple kinds of products involved in TRON ecosystem, including public chain, wallet client, decentralized applications(DAPPs), Etc. We are now ranked 12th on CoinMarketCap. We very much hope that TRX and Trezor-firmware can reach cooperation.

1. This is our [official website](https://tron.network/).

2. The signature algorithm of Tron is ECDSA, and the selected curve is SECP256K1. The private key is a random number, and the public key is a point on the elliptic curve. The generation process is: first generate a random number d as the private key, and then calculate P = d * G as the public key; where G is the base point of the elliptic curve.

3. Sure we have a great front end wallet [Tronlink](https://www.tronlink.org/) and will support Trezor soon.

4. We have a [TRON Developer Hub](https://developers.tron.network/) that includes almost all documents needed for development. This is the [link](https://developers.tron.network/docs/account) of the account address and signature.

5. We have a blockchain explorer [Tronscan](https://tronscan.org/#/).

Starsakary, 2020-10-21:
Are both Trezor One and Trezor T available to add new coins?

tsusanka, 2020-10-27:
We are currently working on our new altcoin support policy and therefore we are not able to provide you with more information. The rules for the new altcoin support should be known in the next upcoming months. Sorry about that, please give us a bit of time to set this up.

WindRing78, 2020-11-19:
Any progress on adding TRON to trezor? TRON is a really good cryptocurrency and Protocol which ranked 16th on CoinMarketCap

stepmes, 2021-08-16:
Thank you, @Starsakary, for submitting this request. I'm now adding the information you provided to our internal list of feature requests related to the support of new projects. This way, it will be included in the prioritization of future work on Suite. We need to run it through the prioritization before we make any decisions. Thanks for understanding!

closed 2021-08-16 by stepmes