trezor/trezor-firmware #1264: Start using calendar versioning

tsusanka, 2020-09-15:
That means YYYY-MM-PATCH. In our case it would be probably YYYY-MM-MODEL-PATCH or similar.

We might not include model into the firmware version, but keep it in the filename. So something like: `trezor-20.8.1-t1.bin` and `trezor-20.8.1-tt.bin` or maybe `trezor-t1-20.8.1.bin` and `trezor-tt-20.8.1.bin`.

We have agreed we want to do this after #24.

prusnak, 2020-09-15:
I prefer to not have model number in the version, it would be very confusing if we announced availability of the 20.8.1.1 and 20.8.1.1 firmware on the same day.

tsusanka, 2020-09-15:
Slush: I would prefer to keep partially current naming to make it more clear. Thus 1.20.8.1 and 2.20.8.1 for model one and T.

prusnak, 2020-11-26:
We have decided we want the following scheme `YY.MM.YY-model{-variant}`, that is:

| | |
|-|-|
| Trezor 1 Legacy Btconly | `20.11.1-t1-btc` |
| Trezor 1 Core | `20.11.1-t1` |
| Trezor T Core | `20.11.1-tt` |
| Trezor T Core Btconly | `20.11.1-tt-btc` |

We need to take care of the following:

**legacy**
- [ ] change version scheme
- [ ] update firmware headers to contain model and add check to bootloader

**core**
- [ ] change version scheme
- [ ] update firmware headers to contain model and add check to bootloader

**python**
- [ ] change firmware update process to reflect the changes

**connect**
- [ ] change firmware update process to reflect the changes

tsusanka, 2020-12-07:
We should start this by investigating whether we need a new `releases.json` (today's format [here](https://github.com/trezor/webwallet-data/blob/master/firmware/1/releases.json)).

prusnak, 2021-02-22:
We could use [JSON Schema](https://json-schema.org) to define the new unified releases.json format.

matejcik, 2022-12-08:
I believe there are no plans to do this. @Hannsek please verify and close if appropriate.

prusnak, 2022-12-08:
I am not super happy that you decided to not use calendar versioning without having proper discussion with me.

There are 0 arguments mentioned here why not to do it.

Can we please discuss this before closing this?

sime, 2022-12-08:
Context, the last discussion was in March: https://www.notion.so/satoshilabs/Model-R-prep-Calendar-FW-versioning-9128f5dfea604c11921467d336a3956b

I chased the action items and the understanding Model R and T versioning had converged. Though it didn't progress further: https://satoshilabs.slack.com/archives/CL1D61PQF/p1664802498000619

matejcik, 2022-12-09:
I do recall from previous discussions that calendar versioning for firmware doesn't really have advantages?
I think the strong argument was sync with Suite, but firmware versions are significantly less frequent than Suite updates, and any clarity gained by "Suite 23.01 with firmware 23.01" is lost on "Suite 23.04 with still firmware 23.01 because there is no 23.02 through 23.04".
Also this loses the semver-like property where we can at least vaguely tell which upgrades are major.

To be clear we can still do this, i was just under the assumption that there _was_ a discussion at some point in the past, and there were no clear advantages in making this change.

matejcik, 2022-12-19:
Closing again after talking to the stakeholders.

Summary:

**Firmware** side this is doable and even relatively straightforward: we would just repurpose version fields in the header and in `Features` message to match up to year/month. It does not _seem_ to pose any challenges to the release process either, but of course we would only know for sure if we actually tried the release.
Two options to do it: (a) keep major version as is and change minor/patch to year/month. (b) change major/minor to year/month, breaking compatibility with software that uses major version to detect model. (such software will need to be fixed for Model R anyway, but some things could be lurking pretty deep in the stack).
see [notion summary](https://www.notion.so/satoshilabs/Firmware-calendar-versioning-936226cae89c4a1ca392d12b37943c61)

**Connect** side this presents some amount of work. Option (b) is basically a no-go because Connect is one of the places with deep hidden references to major-is-model. Option (a) should work seamlessly _in theory_; in practice, we know from past experience that there is Always Something(tm).

**Product** side I'm told there is no strong requirement for this.

In conclusion, this would generate some amount of work with no clearly stated demand or advantage.

closed 2022-12-19 by matejcik