trezor/trezor-firmware #1268: Remove Prompt Always from safety_checks

tsusanka, 2020-09-18:
As discussed on meetings. We want to be strict and then enable if necessary.

andrewkozlik, 2020-09-23:
I think the internal discussion is inclined to keep the `prompt_always` setting, me included. I am summarizing the arguments formulated by @stick and @matejcik, so that they don't get lost:
- avoid crippling the user experience of Casa and anyone that has a different use-case for Trezor than we defined
- "prompt always" mode is in a released firmware already, removing it would do more harm than good
- if we want to do #1260 (skip prevtx streaming), it makes sense that some people with some software would want it always allowed (think Suite + pruned node)

tsusanka, 2021-01-04:
Since the firmware is out already for almost half a year, I agree this could do more harm than good. Closing.

closed 2021-01-04 by tsusanka