trezor/trezor-firmware #1270: Hexadecimal prefix missing from ETH signatures (post 1.8.0)

kargakis, 2020-09-18:
**Describe the bug**
Pre-1.8.0, Ethereum signatures for messages returned by Trezor would include the hexadecimal prefix as part of the signature.
I just updated to 1.9.3, signed a message, and got back a signature without the hex prefix. Is this expected behavior now?

**Firmware version and revision**
1.9.3

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Wallet
 - OS: Linux, also verified on a Windows machine
 - Browser: Firefox 80.0.1, Chrome 85.0.4183.102

kargakis, 2020-09-18:
This may also have always been the case for Trezor. Unfortunately, I don't remember whether I had signed directly via Trezor pre-1.8.0 or most likely combined with MEW, probably the latter. Also, I don't remember having the ability to sign ETH messages directly via the trezor wallet UI pre-1.8, in that case feel free to close the issue.

matejcik, 2020-09-18:
Trezor returns the signature as a binary blob. The way it's presented is fully in the hands of the host software. So if a bug exists, this is not the right repository to report it.

closed 2020-09-18 by matejcik

kargakis, 2020-09-18:
@matejcik what's the right repo for the Trezor wallet UI?

matejcik, 2020-09-18:
There isn't one for the current wallet.trezor.io, i think. You'll need to contact support.
The upcoming Trezor Suite is at https://github.com/trezor/trezor-suite