trezor/trezor-firmware #1272: Add support for EIP-712

danfinlay, 2020-09-18:
Re-opening from [old repo here](https://github.com/trezor/trezor-mcu/issues/401).

EIP 712 is an Ethereum signature format that has two goals:
- More human-readable signatures
- Cheaper to compute on chain signatures

The hope is that this signature format can allow more human-consensual layer-2 interactions on the Ethereum blockchain.

MetaMask has supported this signature format since it was first defined [three years ago](https://github.com/ethereum/EIPs/blob/master/EIPS/eip-712.md). There are a few different variations of it (each with a different RPC suffix: `_v1`, `_v3`, `_v4`, I would recommend focusing on `_v4` if only implementing one).

You can read a bit more about it on [the MetaMask docs](https://docs.metamask.io/guide/signing-data.html#a-brief-history), which do need to be expanded on this topic. We also have some working examples [on this demo repo](https://github.com/danfinlay/js-eth-personal-sign-examples/blob/master/index.js#L373).

I'm happy to answer any questions I can to aid in this development. Many applications would like to adopt this signature scheme, but then users of those applications are currently forced to not use hardware wallets, which is a real shame.

prusnak, 2020-09-18:
Reposting from the old issue:

I suggest you start with Model T first: https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/ethereum/sign_message.py

Then proceed by creating an unit test similar to https://github.com/trezor/trezor-firmware/blob/master/tests/device_tests/test_msg_ethereum_signmessage.py

matejcik, 2020-09-21:
there have been issues about this already:
https://github.com/trezor/trezor-firmware/issues/131
even a WIP PR: https://github.com/trezor/trezor-firmware/pull/148

I'll note that a big problem with EIP712 is that it is very generic and thus difficult to translate to protobuf (and then properly display). This was also the problem with the previous attempt. 

I would suggest not implementing the full format, instead supporting some reasonable subset.

mikec, 2020-12-02:
I'm working on an app that's currently incompatible with Trezor because of this issue. The DAI contract uses EIP-712 for `permit` https://github.com/makerdao/dss/blob/master/src/dai.sol#L120, and many other smart contracts deployed on Ethereum rely on EIP-712.

dmihal, 2021-01-22:
@matejcik I wonder if a "reasonable subset" would be just supporting permit() signatures, since that's probably the most common use of EIP-712.

It should be noted that Dai uses a different signature format than most other apps, which follow the ERC-2612 standard.

matejcik, 2021-01-22:
> I wonder if a "reasonable subset" would be just supporting permit() signatures, since that's probably the most common use of EIP-712.

I'm not watching the space so i'd have to take your word for it. But if you feel that useful, please open a separate issue first (and note wheter you're willing to contribute a PR) so that we can take it into consideration.

mikec, 2021-01-22:
@dmihal @matejcik we're using EIP-712 for a use case that is not `permit()` / ERC-2612. I would think that adding permit() as a subset would actually be more difficult than just adding EIP-712 signing support more generically. I'm willing to contribute a PR, just not sure when I'll be able to get to it.

roshhhyy, 2021-04-03:
hi - I am a Trezor user for years but the lack of eip-712 support is forcing me to have to consider switching to Ledger. I strongly support this integration!

st-lemniscap, 2021-07-31:
Would be good to get a status update from Trezor on support for signTypedData. Is the team even considering this?

discochuck, 2021-08-28:
it appears not

0xklapouchy, 2021-08-28:
What is the status of this issue? More and more users are complaining that they can't use Trezor with our Dapp that uses EIP-712

prusnak, 2021-08-29:
> What is the status of this issue?

There is a pull request for the issue here (https://github.com/trezor/trezor-firmware/pull/1568) which is being reviewed.

barbozadev, 2021-09-28:
> Re-opening from [old repo here](https://github.com/trezor/trezor-mcu/issues/401).
>
> EIP 712 is an Ethereum signature format that has two goals:
>
> * More human-readable signatures
> * Cheaper to compute [...]

Yes guys, we need this to be implemented! Signing in OpenSea doesn't even works with a Trezor One (with latest firmware v1.10.3) connected to MetaMask.
This issue must be critical as OpenSea is used by millions of people.
Screenshots:

[image]
[image]
[image]

You guys can do it!
Thanks!!!

jamesmorgan, 2021-09-28:
It would be really good if this was added, 712 signatures are being used by many and this would enable wider spread use of trezor within the nft space especially

TheArhaam, 2021-11-06:
So strangely I was able to unlock currency on OpenSea earlier using my Trezor One, I had to unlist my NFT and when I tried to put it up for sale again it wouldn't work.

Any idea why or how it worked the first time? 🤔

prusnak, 2021-11-06:
Fixed via https://github.com/trezor/trezor-firmware/pull/1835

Will be released to Trezor Model T in December update

closed 2021-11-06 by prusnak

bonustrack, 2021-11-07:
@prusnak Can you confirm if the model Trezor One will also support EIP-712?

matejcik, 2021-11-07:
Trezor One support is currently not planned.

bonustrack, 2021-11-07:
@matejcik Isn't there some plan to backport the T firmware to Trezor One?

matejcik, 2021-11-08:
Even if this happens, it's unclear if EIP-712 support would be enabled on T1 due to the UI complexity.

prusnak, 2021-11-08:
> Even if this happens, it's unclear if EIP-712 support would be enabled on T1 due to the UI complexity.

RIght. The best way how to interact with EIP-712 contracts is to get Trezor Model T.

bonustrack, 2021-11-08:
Isn't Trezor One the model that is the most used? Feel like EIP712 support will still be a big blocking point if you dont tackle it for model One.

roshhhyy, 2021-11-10:
> > Even if this happens, it's unclear if EIP-712 support would be enabled on T1 due to the UI complexity.
> 
> RIght. The best way how to interact with EIP-712 contracts is to get Trezor Model T.

this isn't a viable solution as many users have polygon nfts trapped on trezor ones and its costing us a lot of money. See sources here:

https://www.reddit.com/r/TREZOR/comments/pwwnb3/polygon_nft_trapped_on_trezor/

https://www.reddit.com/r/TREZOR/comments/o0vxbn/lost_suck_nft_on_trezor/

these are just two examples of the issue, restoring the seed on a different trezor or software wallet is also not viable as it eliminates the entire purpose of having a hardware wallet where the seed is never typed or entered into any computer ever

please support eip712 for the trezor one somehow, it will be a huge service to your most longtime loyal users as the trezor one was the wallet of choice for everyone getting in before 2017

prusnak, 2021-11-11:
> these are just two examples of the issue, restoring the seed on a different trezor or software wallet is also not viable as it eliminates the entire purpose of having a hardware wallet where the seed is never typed or entered into any computer ever

You can restore the seed on Trezor Model T by typing it directly on the device screen.

PabloCastellano, 2021-11-16:
> Even if this happens, it's unclear if EIP-712 support would be enabled on T1 due to the UI complexity.

@matejcik This answer is simply disappointing. You guys are just pushing us, owners of Trezor One, to buy the new model which costs almost 4x. In addition, the website doesn't warn about this lack of support for this model, which is totally the opposite that the buyer expects.

I have no idea about what is the UI complexity you are mentioning. Is it related to reviewing the data you are signing and having such a small screen? If that's the case, I'm sure we can find a balance between having this feature and security.

I'm sure that users still prefer having EIP-712 with a bad UI than otherwise.

My two cents

barbozadev, 2021-11-16:
> @matejcik This answer is simply disappointing. You guys are just pushing us, owners of Trezor One, to buy the new model which costs almost 4x. In addition, the website doesn't warn about this lack [...]

I agree with you! This is a critic issu! ,Please cancel the product for sale or inform buyers that it does not support or will not support EIP 712 signature (For this reason it cannot be used in Opensea, this is serious). But you guys can't say that you don't know yet if it will work on the Trezor Model T, don't use that as an excuse, you have 2 world class products both the Model One and the Model T, the 2 are important. The One model is your first, don't forget that. I hope you think it through for the benefit of the community that uses both models.

roshhhyy, 2021-11-17:
> Even if this happens, it's unclear if EIP-712 support would be enabled on T1 due to the UI complexity.

I think everyone here wouldn't mind having the UI not display the signature information and have to sign the txn relatively blindly in exchange for being able to use the Trezor One for NFTs. I've had a Trezor One since 2017 and a loyal supporter of the product, Ledger figured out EIP712 with the S and X at the same time despite the S having limited memory and UI constraints. The Trezor One should simply support EIP712, UI issues aside.

bonustrack, 2021-11-21:
> I think everyone here wouldn't mind having the UI not display the signature information and have to sign the txn relatively blindly in exchange for being able to use the Trezor One for NFTs. I've [...]

+1

@prusnak Could you explain what it would take to get support on model One? Is the UI the main blocker?

prusnak, 2021-11-21:
> Is the UI the main blocker?

No. Trezor One runs completely different software stack than Trezor Model T.
If you need EIP-712, get a Trezor Model T.

prusnak, 2021-11-21:
> Is the UI the main blocker?

No. Trezor One runs completely different software stack than Trezor Model T.

bonustrack, 2021-11-21:
@prusnak Is it technically possible to add support on that software? Anything we can do to help making this happen?

roshhhyy, 2021-11-22:
these responses are so dismissive - okay yes sure, it runs a different software stack fine. Can said software stack support EIP-712? I feel like saying the software stack makes it IMPOSSIBLE to support EIP712 is a cop out, tons of ETH dapps use EIP712 and other EVM based chains will require signatures on apps to not have the One support EIP712 is basically not supporting ETH itself. 

Trezor support likes to say "The Trezor One doesn't support NFTs" which isn't even true - by not supporting EIP712 the One is not supporting a massive sector of ETH dapps and might as well remove ETH from supported assets. Ledger Nano S figured this out with barely enough memory on the device to have 2-3 asset apps, Trezor One somehow 2 years late is still just dismissing the issue and abandoning ETH + all EVM chains. You're leaving your earliest supporters and users in the dust.

zaneclaes, 2021-11-27:
@prusnak Can we get a beta copy of the `Trezor Model T` firmware update to our users? Our community is being forced to turn away from Trezor (practically everyone is switching to Ledger). I'd love to give them a better option.

prusnak, 2021-11-27:
@zaneclaes Firmware for Trezor Model T with EIP-712 will be officially released in few days - 8th of December.

alisinabh, 2021-11-28:
@prusnak To be clear and honest to us (T1 owners), Are you suggesting that we will not receive this update?

prusnak, 2021-11-28:
> Are you suggesting that we will not receive this update?

I am suggesting you get a Trezor Model T if you need to use EIP-712 now. Everything else is speculation.

bonustrack, 2021-11-28:
> I am suggesting you get a Trezor Model T if you need to use EIP-712 now. Everything else is speculation.

Can you mention in your web store that model One does not support EIP-712?

barbozadev, 2021-11-28:
> Yes guys, we need this to be implemented! Signing in OpenSea doesn't even works with a Trezor One (with latest firmware v1.10.3) connected to MetaMask. This issue must be critical as OpenSea is used [...]

This is the icurrently issue using Trezor model One in Opensea, you guys are pushing us back. This is no take it seriously, must be a critical issue but seems to be low severity for you! Really sad!

prusnak, 2021-11-28:
Thank you for your valuable feedback. We'll discuss this internally. In the meantime I am suggesting everyone getting a Trezor Model T if you need to use EIP-712 now.

prusnak, 2021-12-06:
EIP-712 for Trezor One is being developed here and will be hopefully released in January: https://github.com/trezor/trezor-firmware/pull/1970