trezor/trezor-firmware #1273: Update micropython to 1.13

tsusanka, 2020-09-21:
As @prusnak wrote the interesting changes are:

- PEP 526 has been implemented which adds new syntax for variable annotations, such as x:int = 1
- PEP 572 has also been implemented, which adds the := operator for assignment expressions, such as if x := 1:...
- breaking change: async-for no longer await's on the result of aiter. The aiter method should return an async-iterable object but is not itself awaitable. - does this affect us?

See https://github.com/micropython/micropython/releases/tag/v1.13

prusnak, 2020-09-21:
MicroPython v1.13 + modutimeq.discard patch prepared here https://github.com/trezor/micropython/tree/trezor-v1.13

prusnak, 2020-09-21:
After some fiddling I prepared the update branch here: https://github.com/trezor/trezor-firmware/tree/upy-v1.13

prusnak, 2020-09-23:
Fixed by https://github.com/trezor/trezor-firmware/pull/1280

closed 2020-10-16 by prusnak