trezor/trezor-firmware #1278: Some USB hubs lead to LIBUSB_ERROR_NOTFOUND

tsusanka, 2020-09-23:
Having two Trezors connected to USB hub leads to LIBUSB_ERROR_NOTFOUND on _some_ USB hubs.

How to reproduce:
- Connect T1.
- Connect TT in bootloader.
- Install firmware on TT. Let it reload.
- Run `trezorctl list`.
- Observe the libusb error.

Blocks #946.

## Tested USB hubs:

[Akasa](https://www.alza.cz/akasa-usb-type-c-hub-4-x-usb3-0-type-a-ak-cbca25-18bk-d5834252.htm)
- WORKS!

[Verbatim](https://www.alza.cz/verbatim-usb-c-adapter-usb-3-1-gen-1-usb-3-0x3-hdmi-sdhc-microsdhc-rj45-d5847740.htm)
- libusb error

[D-link](https://www.alza.cz/d-link-dub-h7-d62198.htm)
- libusb error

[Belkin](https://www.alza.cz/belkin-f4u021bt-d2905372.htm)
- libusb error

[TP-Link](https://www.alza.cz/tp-link-uh700-d2417953.htm)
- WORKS!

matejcik, 2020-09-23:
does the same thing happen when going through bridge only?

tsusanka, 2020-09-23:
Tested with the Verbatim one and it fails also for Bridge. In particular it prints the LIBUSB error and nothing else is printed.

If I comment out other transports and keep Bridge only [in here](https://github.com/trezor/trezor-firmware/blob/9a330f34756c7f553c187d007ddce14eea73ca65/python/src/trezorlib/transport/__init__.py#L109) there is no error but Trezor T is not listed.

It seems to me there has to be some weird USB bug in our USB code base.

mcudev, 2021-01-16:
Weird. Is the reproduction case specific to the TT firmware installation while in bootloader mode? Is there a simpler reproduction case? When I watch what happens for my TT device, there is a programmatic USB disconnect when the device reboots and then it registers a a new device in the host system when it boots up again after installing the firmware. Maybe that is somehow interacting with the affected hubs???

Can you reproduce this on a recent Linux like Ubuntu 20.10 (doesn't have to be that version specifically, just something with a relatively up-to-date kernel)?
Can you get the output from (this command will match the Trezor T in bootloader and regular mode): `lsusb --verbose -d 1209:`
Can you get the output from (this will show kernel messages): `journalctl -k | tail -30`

If that is not enough info (probably not), then we can also inspect the USB traffic with wireshark and save it in a capture file:

```
sudo apt-get install wireshark
sudo modprobe usbmon
sudo wireshark &
ls -al /dev/trezor* will show which bus to monitor
for example, bus/usb/003/002 means listen in wireshark on usbmon3
```

tsusanka, 2021-01-17:
> Weird. Is the reproduction case specific to the TT firmware installation while in bootloader mode?

AFAIK yes, I was not able to reproduce it without being in the bootloader. 

> Is there a simpler reproduction case?

No or at least I wasn't able to make one :(.

---

I do not have these hubs at my possession now so I can't test it but I can give it a go later. Or of course if you'd be interested we can send you some of the affected ones.

mcudev, 2021-01-19:
I've got some progress on the low level USB stack config going in another PR (regarding endpoint count and FIFO configs). Let's wait and see if those changes might fix this too. I appreciate the offer of sending me some hubs for testing. I might take you up on that if we can't figure this out easily.

mcudev, 2021-03-09:
the changes that i previously mentioned are being worked as #1517

mcudev, 2021-03-22:
I tried 3 hubs and they all passed the test case described. The T1 and TT devices used these firmwares during the test:

https://data.trezor.io/firmware/1/trezor-1.9.4.bin

https://data.trezor.io/firmware/2/trezor-2.3.6.bin

1. https://manhattanproducts.us/products/manhattan-en-hi-speed-usb-20-ultra-hub-161039

```
journalctl -ke
  high-speed USB
  idVendor=1a40, idProduct=0201, bcdDevice= 1.00

lsusb -t
  If 0, Class=Hub, Driver=hub/7p, 480M
```

2. https://www.sabrent.com/product/HB-UMLS/4-port-usb-2-0-hub-power-switches-black/

```
journalctl -ke
  high-speed USB
  idVendor=1a40, idProduct=0101, bcdDevice= 1.11

lsusb -t
  If 0, Class=Hub, Driver=hub/4p, 480M
```

3. https://www.cnet.com/products/rosewill-rhb-330-hub-7-ports/

```
journalctl -ke
  high-speed USB
  idVendor=1a40, idProduct=0201, bcdDevice= 1.00

lsusb -t
  If 0, Class=Hub, Driver=hub/7p, 480M
```

tsusanka, 2021-03-30:
Hm :thinking:. I do not have these three hubs that I have mentioned anymore but I'll reorder one of them to retest.

tsusanka, 2021-04-09:
Hm I just tried the very same D-link mentioned above and I am not able to reproduce it. And even I tried some older firmwares. Not really sure what we have learned here, but I guess I am closing..

closed 2021-04-13 by tsusanka