trezor/trezor-firmware #1286: Send specific ButtonRequest for specific device settings

tsusanka, 2020-09-29:
Suite team is quite sad we do not send particular ButtonRequest for: 

- PIN 
- Labeling 
- Device name 
- Display orientation

Although labeling might be more complex I believe sending particular button request for the others is no problem? And it should be backwards compatible because it only adds new ones?

I know we have #35 but this is a three-liner IMHO and will improve Suite so might be worth doing asap?

matejcik, 2020-09-29:
there's `ButtonRequestType.PinEntry`

As for the rest, that's super specific. We also don't have specific buttonrequests for setting auto-lock delay, passphrase, homescreen, and safety checks.

As long as it's just these three (or four), we can do this. But let's not make special buttonrequests for every kind of dialog. That's what #35 is for.

We could also do #35 partially and introduce `ButtonRequest.type` field for these?

matejcik, 2020-09-29:
...isn't "labeling" and "device name" the same thing?

tsusanka, 2020-10-08:
@matejcik is right. Let's wait for #35 to do this properly.

closed 2020-10-08 by tsusanka