trezor/trezor-firmware #1287: Not able to send timelocked transactions

bosomt, 2020-09-29:
**Describe the bug**

Not able to send timelocked transactions for years. 
See issue that I sent year and half ago https://github.com/satoshilabs/mytrezor/issues/736
All firmwares including T1 and TT.
Same problem in suite and wallet and in send raw transaction tool.

**Firmware version and revision**
1.9.2 commit 0xcde8f31ec2ddcb7d35e36edbcf8a71dda983a9ea,

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: current wallet /commit 0ce221e99a33c1a8c728f7dbea79d85ae7851ac3/ and suite 
 - OS: Nixos
 - Browser Chromium 85
 - Version 85

**To Reproduce**
Steps to reproduce the behavior:
1. load wallet or suite
2. use locktime function
3. try to send transaction
4. See error [account] Failed to send transaction",{"message":"-26: non-final"},

**Expected behavior**
You are able to send locktimed transaction

**Screenshots**

[image]

[image]

[image]

matejcik, 2020-09-29:
this does not look like a firmware problem, and ISTM it is expected behavior, despite the error message being confusing

transacitions where `lock_time` > `current block height` are going to be refused by the network. that is the whole point of `lock_time`

try using locktime lower than current block height

closed 2020-09-29 by matejcik

bosomt, 2020-09-29:
worked fined when i used locktime that already happened `  "locktime": 650489,` 
https://btc3.trezor.io/tx/1edcddd0991df80843313e06b9a5a5218ce7897ec0d63e348894d980b4065a69

tsusanka, 2020-09-29:
Let's improve the error message then. Do we know where from this error comes? Maybe blockbook?

matejcik, 2020-09-29:
pretty sure it's straight out of bitcoin core

i'm not sure where it should be handled, but my intuition is that it should be in Wallet/Suite

tsusanka, 2020-09-30:
I've opened issue in Suite https://github.com/trezor/trezor-suite/issues/2463.

MichielBruijn, 2022-05-14:
> this does not look like a firmware problem, and ISTM it is expected behavior, despite the error message being confusing
>
> transacitions where `lock_time` > `current block height` are going to be [...]

What's the point of locktime lower than current block height?
I want to understand

matejcik, 2022-05-15:
> What's the point of locktime lower than current block height?

none except for testing that transactions with locktime are signed correctly.

MichielBruijn, 2022-05-15:
> > What's the point of locktime lower than current block height?
> 
> none except for testing that transactions with locktime are signed correctly.

So why is locktime higher than current block height not working and gives errors?

matejcik, 2022-05-16:
i suggest you read up on what "lock time" means and does, and it will hopefully clear up your confusion

MichielBruijn, 2022-05-16:
[This page](https://wiki.trezor.io/Locktime) says:

_Locktime is an optional part of Bitcoin. It sets the earliest time a transaction can be mined into a block. It is possible to use locktime with Trezor to make sure that a transaction is locked until a specific block height or a point in time._

So, locktime > current block height is how it's suposed to work but it's not possible to broadcast the transaction until current block height > locktime.
So, the broadcast button in trezor suite is confusing when using locktime > current block height.

I was under the impression that locktime was a feature to "lock" the coins on the chain until a specific point in (block) time where they come available again.

prusnak, 2022-05-16:
You can't broadcast a transaction when timelock has not been reached.

sime, 2022-05-16:
@MichielBruijn You might be thinking of OP_CHECKLOCKTIMEVERIFY and OP_CHECKSEQUENCEVERIFY. Trezor doesn't support these, and they behave differently than LockTime.

MichielBruijn, 2022-05-16:
Ah, that explains. I was indeed thinking about OP_CHECKLOCKTIMEVERIFY aka OP_HODL
Thanks for clearing up.
Do you know if those features are going to be implemented into Trezor in the futre?