trezor/trezor-firmware #1290: On-demand locking

dekoza, 2020-09-30:
**Is your feature request related to a problem? Please describe.**
I'm frustrated with Trezor locking after several minutes after latest firmware update. I like this auto-lock feature as it increases the security but I'd like to be able to configure this behavior to my needs.

**Describe the solution you'd like**
I'd like to have an option to disable auto-lock or extend the time of inactivity that locks Trezor. Additionally I'd like to lock the device manually or send a lock-request via USB (for example when locking my workstation I'd like my Trezor to be locked automatically).

tsusanka, 2020-09-30:
Thanks for the feedback! Currently you can change the autolock delay value via trezorctl: `trezorctl set auto-lock-delay 10h` (for 10 hours). More info on how to get `trezorctl` can be found on [our wiki](https://wiki.trezor.io/Using_trezorctl_commands_with_Trezor).

That is of course a bit cumbersome, we plan to add this setting to our new upcoming Suite wallet soon. See https://github.com/trezor/trezor-suite/issues/1946.

tsusanka, 2020-09-30:
Since there is no action left at the moment I am closing but feel free to ask more questions if necessary.

closed 2020-09-30 by tsusanka

dekoza, 2020-09-30:
I totally forgot about `trezorctl`, sorry :sweat_smile: This solves my "lock on workstation lock" issue since I can set the lock delay to 1s (and then set it back to 10h) which is good enough. Thank you!

matejcik, 2020-10-01:
you can't set to 1s, 10s is minimum

you can, however, `trezorctl clear-session` which insta-locks