trezor/trezor-firmware #1293: Support for verifiable manual entropy

jimmysong, 2020-09-30:
Compromised random number generation is one of the main security threats to any hardware device. This is a way to verifiably create a BIP39 HD seed that would be a useful feature to have.

Currently, there are only two choices when loading the hardware with a seed:

1. Have the device generate it
2. Restore a seed (essentially load your own seed, probably generated on another device, into the device)

I would like a third option that lets me manually create my own seed that passes a checksum. Supposing that I have some way to generate manual entropy (dice, cards, etc), I would like to input a bunch of words directly into the device, but have the device give me the possible checksum words. Currently, 12, 18 and 24 word seeds require 4, 6 and 8 bits of checksum respectively. As the last word contains 11 bits of information, this means that for 24 words, I should have the option of picking my first 23 words and seeing the 8 possible last words (11 bits - 8 bits is 3 bits, which is 8 permutations). For 18, this is 32 possible last words and for 12, this is 128 possible last words. A number in front of each word would be very helpful for the manual entropy aspect. That way, I can roll dice or flip coins or something to choose that last word.

Essentially, this is very similar to restore-a-seed, but giving me a way to generate my own seed manually without use of any other device. Ideally, this process would be done in conjunction with a paper or metal backup. From a security standpoint, This would eliminate the dependency on the Trezor's RNG.

This would also have to be doable without the Trezor being plugged into a computer when that feature is ready.

tsusanka, 2020-10-01:
We are considering doing the diceware/cointoss directly in Trezor https://github.com/trezor/trezor-firmware/issues/23.

jimmysong, 2020-10-01:
> We are considering doing the diceware/cointoss directly in Trezor #23.

That protects very narrowly against the RNG being bad. It's hard for a user to know whether the dice throws correspond to the seed that's generated without another device. The point is to be able to verify the seed phrase and not have to use another device.

dpc, 2020-12-07:
Copying from #1381 :

> **Is your feature request related to a problem? Please describe.**
> I was looking at the https://twitter.com/kallerosenbaum/status/1335185904667942920 where the author expressed frustration with the checksum being an obstacle to generating seeds entirely offline using dice.
> 
> **Describe the solution you'd like**
> After a bit of thinking it seems to me that the best approach would be for the wallet to just tell the user what the last word should actually be (if it was incorrect). This would allow people to just generate all words randomly without even being aware of checksum, and then fix the last word when they are trying the seed for the first time.
> 
> This even works if the user already did multiple copies of the seed etc. Any time they enter the seed, the checksum suggested is going to be deterministic.
>
> Possibly the wallet should require the user start from scratch with the fixed word, just to make sure the wrong checksum wasn't just a typo.