trezor/trezor-firmware #1294: Show fee rate on Trezor

andrewkozlik, 2020-10-01:
It might be convenient for users to show them the fee rate on the singing confirmation screen. On the TT we have one spare line at the bottom, where we can put this information:

[image]

Not sure how to do this on the T1 though:

[image]

We could get rid of the "from your wallet?", so we would have

```
Confirm sending:
0.01610436 TEST
including fee:
0.00005436 TEST
(83 sat/B)
```

The implementation would be pretty easy, but I would like to double-check the way we calculate the transaction weight in Trezor and in Wallet/Suite to ensure that the computation is the same, so that we avoid showing a different fee rate on Trezor in some odd cases. The differences in computation would probably be minimal, boiling down to the things which are unknown before the transaction is signed, i.e. the signature length - do we take the average length or the upper bound?

closed 2022-05-25 by matejcik

sime, 2022-06-20:
@STew790 Since you tested the Suite side of things, could you also test firmware as well?

STew790, 2022-06-24:
**QA OK**

T1:
[image]
TT:
[image]

**Info:**
 - Suite version: web 22.7.0 (a92092adee0fc61f4bc2c8a404f40e47323c1450)
 - Browser: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/102.0.5005.115 Safari/537.36
 - OS: Linux x86_64
 - Screen: 1920x1080
 - Device: model 1 1.11.2 regular (revision 468bcfd61a870dc11ca6c5e8555947bf05de4a57)
 - Device: model T 2.5.2 regular (revision 468bcfd61a870dc11ca6c5e8555947bf05de4a57)
 - Transport: bridge 2.0.31