trezor/trezor-firmware #1296: [2.3.1+] build_tx.py no longer produces viable JSON for btc sign-tx [Encountered invalid prev_hash]

brianddk, 2020-10-03:
**Describe the bug**
In release 0.12.0, 0.12.1, and 0.12.2, `build_tx.py`, given the same input will produce the same `txn.json` file.  But this `txn.json` file will only serve as viable input for the 2.3.0 FW.  All FW beyond 2.3.0 will throw the errror:

```
Error: ProcessError: Encountered invalid prev_hash
```

**Firmware version and revision**
Firmware 2.3.1, 2.3.2 and 2.3.3 when used with TrezorCTL 0.12.0, 0.12.1, or 0.12.2.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: TrezorCTL
 - OS: Windows 10
 - Python: 3.7.1

**To Reproduce**
This is an attempt to reproduce testnet TXN [43d273d3caf41759ad843474f960fbf80ff2ec961135d018b61e9fab3ad1fc06](https://tbtc1.trezor.io/tx/43d273d3caf41759ad843474f960fbf80ff2ec961135d018b61e9fab3ad1fc06) as a test.

Steps to reproduce the behavior:
1. Install TrezorCTL (`pip install trezor`) version 0.12.0, 0.12.1 or 0.12.2
2. Install FW 2.3.0
3. Grab the `input.sh` script (see below)
4. `input.sh | python tools/build_tx.py | trezorctl btc sign-tx -`
5. Observe the ***successful*** test case
6. Install FW 2.3.1 or 2.3.2 or 2.3.3
7. `input.sh | python tools/build_tx.py | trezorctl btc sign-tx -`
8. Observe ***failure*** test case: `Error: ProcessError: Encountered invalid prev_hash`

#### input.sh
```shell
#!/bin/bash
sleep 1;echo "Testnet"
sleep 1;echo "tbtc1.trezor.io"
sleep 1;echo "e294c4c172c3d87991b0369e45d6af8584be92914d01e3060fad1ed31d12ff00:0"
sleep 1;echo "m/84'/1'/0'/0/0"
sleep 1;echo "4294967293"
sleep 1;echo "segwit"
sleep 1;echo ""
sleep 1;echo "2MsiAgG5LVDmnmJUPnYaCeQnARWGbGSVnr3"
sleep 1;echo "10000000"
sleep 1;echo "tb1q9l0rk0gkgn73d0gc57qn3t3cwvucaj3h8wtrlu"
sleep 1;echo "20000000"
sleep 1;echo "tb1qejqxwzfld7zr6mf7ygqy5s5se5xq7vmt96jk9x"
sleep 1;echo "99999694"
sleep 1;echo ""
sleep 1;echo ""
sleep 1;echo "2"
sleep 1;echo "0"
```

**Expected behavior**
Expect `python tools/build_tx.py | trezorctl btc sign-tx -` to always work when running the latest firmware and latest TrezorCTL

**Additional context**
The code path for the error is 

* [core/src/apps/bitcoin/sign_tx/bitcoin.py: line 453](https://github.com/trezor/trezor-firmware/blob/core/v2.3.3/core/src/apps/bitcoin/sign_tx/bitcoin.py#L453)

I suspect it is related to the refactor ([commit 445f56d](https://github.com/trezor/trezor-firmware/blob/445f56d387471d6d6c99fa01c8d5e22b6a21868e/core/src/apps/bitcoin/sign_tx/bitcoin.py)) that happened between 2.3.0 and 2.3.1.

matejcik, 2020-10-06:
this was fixed via #1257

closed 2020-10-06 by matejcik

brianddk, 2020-10-06:
@matejcik , Thx, it does fix it but installing the python libraries from `master` creates another problem related to the new calling conventions for `SignTx` class creation.  I made PR #1300 to fix it, if that is the path you want to take.