trezor/trezor-firmware #1302: Add pagination for passphrase

NIYAK36, 2020-10-07:
Option for reviewing a whole passphrase on Trezor display.

tsusanka, 2020-10-07:
In case the passphrase is longer than the keyboard input.

prusnak, 2020-10-07:
More detail: currently on Model T we show passphrase on display after it has been entered on the host. This does not happen when the passphrase is entered on the device. However, there are people who would like to check their very long passphrase.

Straightforward solution would be to show the confirm passphrase dialog in every case (both for entering on host and entering on the device). I think that's what we should do, as it also reduces confusion why sometimes the dialog is shown and sometimes not (if person changes the input method).

tsusanka, 2020-10-20:
This is what entering longer passphrase looks like at the moment:

[image]

tsusanka, 2020-10-29:
> Straightforward solution would be to show the confirm passphrase dialog in every case (both for entering on host and entering on the device). I think that's what we should do, as it also reduces confusion why sometimes the dialog is shown and sometimes not (if person changes the input method).

Yes, let's do that now.

My1, 2020-11-01:
seems like a good idea, but by the way what's that red dot in the simulated screencap?

prusnak, 2020-11-01:
@my1 it's there when you run the device/emulator with debugging enabled

tsusanka, 2020-11-16:
Thanks a lot @NIYAK36 for the feedback.

Upon discussion in our Product team we have decided we do not want to change this. The reason is that in Suite (our new wallet) you are asked to repeat the passphrase in case you enter a passphrase which is "empty" (the passphrase leads to a wallet containing nothing).

To elaborate, in case you enter passphrase and the wallet is empty you will see this dialog:

[image]

And Trezor will ask you to enter the passphrase again. At this moment you either enter the same passphrase and therefore it is the correct one (entering the passphrase twice incorrectly while having the device in your hands and watching what you are entering seems extremely unlikely) or you enter different one and the process starts over.

So we believe there is no need for another dialog and we want to keep the UI as clean as possible. We do not want to arrive to the "installer on Windows" workflow where you just click Next without knowing what you are doing :).

closed 2020-11-16 by tsusanka