trezor/trezor-firmware #1307: Run environment once a week

tsusanka, 2020-10-14:
Using schedules. What is annoying is that we can't have schuduled and manual job at the same time AFAIK. Are we okay to ditch the manual option? We can use the same trick as we do with hw tests: run the job for branches prefixed with `env/` or similar.

tsusanka, 2020-10-15:
Let's have two jobs one for manual, one for schedules.

vdovhanych, 2021-05-28:
Not really. You can go ahead and run the scheduled job manually from the "Schedules" page in the project ci/cd settings. Shown [here](https://docs.gitlab.com/ee/ci/pipelines/schedules.html#running-manually).

prusnak, 2021-10-07:
> Moved to [satoshilabs/devops#48](https://github.com/satoshilabs/devops/issues/48).

Should we close this issue, then?