trezor/trezor-firmware #1309: Add support for Ergo tokens

mark-stopka, 2020-10-14:
**Is your feature request related to a problem? Please describe.**
I am frustrated I cannot store my Ergo tokens on my Trezor Model T.

**Describe the solution you'd like**
I would like a combination of Yoroi (which already supports Trezor Model T for Cardano & Ergo) and Trezor Model T to be bale to protect my Ergo tokens.

**Describe alternatives you've considered**
Yoroi paper wallet, not secure for common use.

**Additional context**
I have discussed with Alexander Chepurnoy of Ergo, and he is looking for paid contractor, would SatoshiLabs be able to add such support, or should we contact Vacuum Labs who wrote the Cardano integration?

raymonddurk, 2020-10-14:
The team has said they will only support top 30 coins by market cap and it is currently 380. Is this a fork of Cardano?

mmahut, 2020-10-14:
It is not a fork of Cardano. Also, almost no chance this would get merged.

mark-stopka, 2020-10-14:
@raymonddurk , it is not a a fork of Cardano in any way (Ergo is in Scala, Cardano in Haskell for instance), it is just done by a former IOG researcher. Anyway, I have created this ticket merely for a reference, I have contacted @prusnak and @MichalPetro out of band via e-mail, providing more visibility into this matter.

tripflex, 2021-08-15:
Agreed, after eth this coin will start to pop, come on guys get with it!

stefanobieler, 2021-09-11:
YES PLEASE! I just got my Trezor wallet. kind of disappointed that this coin is not included. please add support for this coin.

hynek-jina, 2022-04-22:
@mark-stopka 
We do not plan to add this coin anytime soon. If the situation changes, we will reopen the issue.

closed 2022-07-21 by hynek-jina