trezor/trezor-firmware #1310: Add XTZ support to Trezor One

shekhar-shubhendu, 2020-10-14:
**Describe the solution you'd like**
Consider providing support for XTZ in Trezor one

**Describe alternatives you've considered**
With lack of this feature, only alternative might be here to either get a Ledger wallet which comes at the same cost as Trezor One or upgrade to Trezor T model which is costly (at least for beginners and devs).

**Additional context**
Tezos appears in TOP30 and it should be part of Trezor One as it is still sold (although is now "legacy") and a popular choice.

prusnak, 2020-10-15:
We are porting Model T firmware to Model One. Once this happens, Model One will support coins from Model T (including XTZ).

closed 2020-10-15 by prusnak

shekhar-shubhendu, 2020-10-15:
@prusnak that would be amazing :)
Thanks for the answer.