trezor/trezor-firmware #1311: Any reason why the PCB design files are not shared and only a PNG of the schematic?

iGetSchwifty, 2020-10-14:
When looking for the schematic all I find is a PNG instead of actual PCB design files. Such as (brd & sch for EAGLE) or the equivalent for KiCad.

As someone who enjoys open source and especially open source HW projects. I find it nice to have access to these files so that I do not need to recreate everything by hand if I wanted to mess around with an idea.

Im sure there is a reason why a PNG was uploaded instead of the design files. There is a human element step in order to recreate those PNG design into the actual design for say KiCad or EAGLE. In other words, the schematic needs to be placed onto the board still with only exposing the PNG file.

I am sure Ill be told why the PNG was only uploaded and that is okay as I am still grateful for that to be shared. The schematic is really the meat and potatoes and everything else can be recreated in KiCad or EAGLE with a little manual labor, which I am capable of doing but if I dont ask then I did not try. Im okay with saying no and closing this ticket if that is where the answer lay.

prusnak, 2020-10-15:
brd/sch files for eagle are here: https://github.com/trezor/trezor-hardware/tree/master/electronics

closed 2020-10-15 by prusnak

jeffrey-cripe, 2020-10-18:
Thank you and I'm sorry I missed those before!