trezor/trezor-firmware #1319: EnergyWeb chain and EW testnet Volta not supported on Trezor One

bartcoelus, 2020-10-20:
**Describe the bug**
Trezor One does not allow transactions to sign using Trezor One. Error: Invalid chainID. The first-gen Trezor (model T) does not show that same error. Also would be good to add EnergyWeb and Volta as officially supported blockchain. 

**Firmware version and revision**
2.3.x

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Metamask / Mycrypto.com
 - OS: macOS and Windows
 - Browser: Chrome
 - Version: all (I tested with Version 86.0.4240.80 (Official Build) (x86_64))

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Trezor One to metamask/Mycrypto
2. Connect wallet
3. Initiate transaction
4. "invalid chainID" error

**Expected behavior**
The transaction should sign - works on Trezor T 

**Screenshots**
NA

**Additional context**
Add any other context about the problem here.
For VOLTA:
https://energyweb.atlassian.net/wiki/spaces/EWF/pages/703201459/Volta+Connecting+to+Remote+RPC+and+Metamask
Go to Networks and click on "Custom RPC". Now go to "Network" and click on "Add Network". Now you can add https://volta-rpc.energyweb.org, enter the chainID of Volta: "73799", as a Symbol put "VT" and "Volta RPC" or something similar as the name.

For EnergyWeb chain:
https://energyweb.atlassian.net/wiki/spaces/EWF/pages/723681315/Connecting+via+Remote+RPC#:~:text=Go%20to%20Networks%20and%20click,a%20Symbol%20put%20%22EWT%22
Go to Networks and click on "Custom RPC". Now go to "Network" and click on "Add Network". Now you can add "EnergyWeb RPC" or something similar as the name, https://rpc.energyweb.org/ as the RPC URL, enter the chainID of the Energy Web Chain: "246", as a Symbol put "EWT"

matejcik, 2020-10-20:
EnergyWeb and Volta are not listed as officially supported, so it's not a bug if they don't work.
(Also I think you have this the wrong way around? IIRC it does not work on TT but works on One)

EWT/Volta implementation was piggy-backing on Ethereum support and showing transactions with "ETH" currency symbol, which we consider a potential security risk, and so it is no longer allowed.
We will accept contributions to add proper EWC/Volta support, see readme in [common/defs](https://github.com/trezor/trezor-firmware/tree/master/common/defs) subdirectory

closed 2020-10-20 by matejcik

bartcoelus, 2020-10-20:
Hi @matejcik  - thanks for taking a look so quickly! Yes, you're right, it works on the One, not on the T. Also it's good to see a confirmation why it's not working any longer, costed me hours of debugtime to figure out what was wrong with it :) 
Thanks a lot for pointing me to the correct procedure to add EWC/Volta - we're on it!

ngyam, 2020-10-21:
@matejcik Just to confirm, which derivation path results in the same addresses on both chains? Like on Trezor One. m/44'/60'/0'/0? ("slip44": 60)One of the main reasons it is important for us is we have a token bridge with which you send/receive to the same address between ETH <-> EWT

matejcik, 2020-10-21:
Ethereum network addresses are not affected by the chain id, so it should be possible to use Ethereum's derivation path `m/44'/60'/*`

I'll need to take cross-signing into consideration when continuing on #1184, as long as the chain id is different this should be safe. But you will still need to specify a valid SLIP44 id in the definition.

How exactly does this token bridge work?

ngyam, 2020-10-22:
chain ID would be different of course.

The bridge is simply the [POA bridge](https://docs.tokenbridge.net/about-tokenbridge/features), we have it running on https://bridge.energyweb.org. It's a lock/mint, burn/unlock bridge, native EWT tokens can be swapped to ERC20 tokens called EWTB on the mainnet which are pegged 1:1. So you can trade on Uniswap and other DEXes.

Both EnergyWebChain and Eth mainnet are EVM chains, so the crypto for addresses are the same, based on secp256k1.

You send native EWT to the bridge contract on Energy Web Chain, it gets locked, and the equal amount of ERC20 token gets minted on the mainnet. The bridge sends a tx  to mint the token to the same address you sent the EWT from cause the address exists on both chains and it assumes you hold the private key for it already.

So it is important for the end users to:
- see/access the same addresses on both EWC/mainnet
- to be able to use Trezor with Metamask and check/sign/send tx on both chains with the same addesses, because the bridge UI works with Metamask

For this reason we always recommended everyone to use Trezor Ones which became our gold standard of hw wallets.