trezor/trezor-firmware #1322: some settings stay on after wipe until device is rebooted

matejcik, 2020-10-22:
e.g., device rotation and auto-lock delay will stay with the same configuration, because the value is erased from storage but remains in effect in separate objects.

Ideal solution would be to reboot the device after wipe, but that is impractical from a technical standpoint.

As an interim solution, we could introduce something like `apply_settings_from_storage()` which would be called both at end of `ApplySettings` handler and `WipeDevice` handler

matejcik, 2020-10-22:
experimental flags are probably also affected

closed 2020-11-05 by matejcik

bosomt, 2020-12-21:
**QA OK**

i was not able to replicate it anymore, rotation and autolock is forgotten even when device is not reconnected
 
Nixos 
Suite be4e3a36f7dbe15b2b203f66de11edc50a6665b8
Device: fw-regular-2.3.5-889a3699

commit c9af066f (2020-10-26): fix(core): make sure run-time settings are reset after wipe (fixes #1322)
commit 526d1051 (2020-10-26): fix(core): make sure run-time settings are reset after wipe (fixes #1322)
commit 20ec6cc1 (2020-10-27): fix(core): make sure run-time settings are reset after wipe (fixes #1322)
commit e83d5485 (2020-11-05): fix(core): make sure run-time settings are reset after wipe (fixes #1322)